ibmtjbot/tjbot #131: Using normal RGB Led

gqferreira, 2019-02-15:
How to use common RGB LED? Do I need resistors? What is the scheme for soldering and bonding in the GPIO? I can not find NeoPixel to buy.

Thank you!

jweisz, 2019-02-15:
The Neopixel is listed as in stock on Adafruit:
https://www.adafruit.com/product/1734

Using LEDs other than the Neopixel is not officially supported, but I'm sure it's possible to do. I don't have good guidance on how, though, since I've only used the Neopixel.

gqferreira, 2019-02-15:
> The Neopixel is listed as in stock on Adafruit:
> https://www.adafruit.com/product/1734
> 
> Using LEDs other than the Neopixel is not officially supported, but I'm sure it's possible to do. I don't have good guidance on how, though, since I've only used the Neopixel.

Yes but the shipping is the problem for me:
[image]

mkhuthir, 2019-03-29:
TJBOT library is using **rpi_ws281x** library to control color and brightness of the **Neopixel LED** through **BCM 18 (PWM0)** pin on Raspberry PI. You will not be able to use a simple LED or RGB LED to replace the Neopixel unless you change TJBOT library to support the LED or RGB LED instead of Neopixel.

jweisz, 2020-02-03:
Closing due to inactivity.

closed 2020-02-03 by jweisz