ibmtjbot/tjbot #133: Unable to complete "Build A Talking Robot"

zacaintmyname, 2019-03-10:
Hi, I am new to this field.
I was playing around with TJBot and I have run into a problem.

Would you be able to let me know what went wrong? 

Am using the tutorial from https://github.com/ibmtjbot/tjbot/blob/master/recipes/conversation/README.md.

Below is what I experience when I verbally ask the bot "Watson, introduce yourself".
info: TJBot heard: Watson introduce yourself 
error: the assistant service returned an error. message=Missing required parameters: workspace_id, stack=Error: Missing required parameters: workspace_id
    at Object.getMissingParams (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/helper.js:95:11)
    at AssistantV1.message (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/assistant/v1.js:96:38)
    at TJBot.converse (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:719:21)
    at /home/pi/Desktop/tjbot/recipes/conversation/conversation.js:53:12
    at RecognizeStream.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:798:13)
    at RecognizeStream.emit (events.js:180:13)
    at addChunk (_stream_readable.js:274:12)
    at readableAddChunk (_stream_readable.js:261:11)
    at RecognizeStream.Readable.push (_stream_readable.js:218:10)
    at /home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:306:34
    at Array.forEach (<anonymous>)
    at W3CWebSocket.socket.onmessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:299:34)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at WebSocketConnection.emit (events.js:180:13)

Will anyone be able to help me out? Thank you!

jweisz, 2019-03-11:
It looks like the Assistant service doesn't know which workspace to use:
> Missing required parameters: workspace_id

Did you do the following?

> Import the workspace-sample.json file into the Watson Assistant service and note the workspace ID.

> Make a copy the default configuration file and update it with the Watson service credentials and the conversation workspace ID.

```
$ cp config.default.js config.js
$ nano config.js
<enter your credentials and the conversation workspace ID in the specified places>
```

zacaintmyname, 2019-03-12:
Hi Sir,

Thank you for the reply. Indeed I have added the workspace ID (which I believe is a 'skill' under Watson Assistant)

Unfortunately I am now facing another error prompt. 

verbose: TJBot initializing microphone
verbose: TJBot initializing LED
verbose: TJBot initializing servo motor on PIN 7
2019-03-12 16:23:58 initInitialise: Can't lock /var/run/pigpio.pid
/home/pi/Desktop/tjbot/recipes/conversation/node_modules/pigpio/pigpio.js:11
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/pigpio/pigpio.js:11:12)
    at new Gpio (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/pigpio/pigpio.js:25:3)
    at TJBot._setupServo (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:304:19)
    at TJBot.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:71:18)
    at Array.forEach (<anonymous>)
    at new TJBot (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:56:14)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/conversation.js:36:10)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:695:10)
    at startup (internal/bootstrap/node.js:201:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:516:3)

zacaintmyname, 2019-03-12:
Sorry managed to troubleshoot this already! Works now! 
Mainly uncommenting and commenting the user/apikey.

Thank you!

closed 2019-03-12 by zacaintmyname