ibmtjbot/tjbot #134: How to update tjConfig?

zacaintmyname, 2019-03-13:
Hi Team,

I came across this and I have the exact same problems as EsthervanLuit initially faced.
How do you update "tjConfig". 

Sorry new to coding, hope you do not mind! 

------------
Hi @EsthervanLuit ,

I do agree with you that its an audio problem (the app uses deviceid 0,0, but should use 2,0). 
With the tjbot library, you can specify the `speakerDeviceId` .. and in your case, this should be set to 2,0. I would say we update `tjConfig` to 

```
var tjConfig = {
    verboseLogging: true,
    speakerDeviceId: 'plughw:2,0',
    microphoneDeviceId: 'plughw:1,0',
        robot: {
                name: 'TJ',
                homophones: ['DJ','Deejay','Teejay'],
                gender: 'female'
    },
speak: {
    language: 'en-US', // see TJBot.prototype.languages.speak
    voice: undefined, // use a specific voice; if undefined, a voice is chosen based on robot.gender and speak.language
    speakerDeviceId: "plughw:2,0" // plugged-in USB card 1, device 0; see aplay -l for a list of playback devices
  }
};
```

For the full list of parameters you can configure with the TJBot library, please see [here](https://github.com/ibmtjbot/tjbotlib#usage).  (If the above does not work, one final thing you might have to do is roll back changes to you might have made to alsa.conf and asound.conf)

-V.

_Originally posted by @victordibia in https://github.com/ibmtjbot/tjbot/issues/63#issuecomment-331750652_

maryamashoori, 2019-03-13:
Hi there, you need to make a copy of the default configuration file and update the text. Here is how to do it from the command line:

`$ cp config.default.js config.js`

and then 
`$ nano config.js`

zacaintmyname, 2019-03-15:
@maryamashoori  Thank you for the reply! 

But $ cp config.default.js config.js brings me to:
```
// set up TJBot's configuration

exports.tjConfig = {
    log: {   level: 'verbose'    },
    robot: {   name: 'tee jay bot'  }
};
```

I was looking for the **var tjconfig** part.
`var tjConfig = {
    verboseLogging: true,
    speakerDeviceId: 'plughw:2,0',
    microphoneDeviceId: 'plughw:1,0',
        robot: {
                name: 'TJ',
                homophones: ['DJ','Deejay','Teejay'],
                gender: 'female'
    },
speak: {
    language: 'en-US', // see TJBot.prototype.languages.speak
    voice: undefined, // use a specific voice; if undefined, a voice is chosen based on robot.gender and speak.language
    speakerDeviceId: "plughw:2,0" // plugged-in USB card 1, device 0; see aplay -l for a list of playback devices
  }
};`

Is it the same? Thanks for helping!

closed 2019-04-09 by zacaintmyname