ibmtjbot/tjbot #135: Error in Conversation Recipe

ryanomor, 2019-03-26:
After entering my credentials in my `config.js` file, I tried running the conversation recipe using `sudo node conversation.js` but I keep getting this error:
```
verbose: TJBot initializing microphone
verbose: TJBot initializing LED
verbose: TJBot initializing servo motor on PIN 7
verbose: TJBot initializing assistant service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
verbose: TJBot initializing tone_analyzer service
verbose: TJBot initializing visual_recognition service
/home/pi/Documents/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:542
            throw new Error(
            ^

Error: Missing authentication credentials for tone_analyzer service: username/password or apikey are required.
    at TJBot._createServiceAPI (/home/pi/Documents/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:542:19)
    at new TJBot (/home/pi/Documents/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:116:18)
    at Object.<anonymous> (/home/pi/Documents/tjbot/recipes/conversation/conversation.js:36:10)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:695:10)
    at startup (internal/bootstrap/node.js:201:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:516:3)
```

jweisz, 2019-03-26:
It looks like you didn't put in credentials for the tone analyzer service:

> Error: Missing authentication credentials for tone_analyzer service: username/password or apikey are required.

jweisz, 2019-03-26:
Hm. On the surface it looks like you didn't put in your authentication credentials for the tone analyzer service:

> Error: Missing authentication credentials for tone_analyzer service: username/password or apikey are required.

But that said, the Conversation recipe doesn't require tone analyzer. It does require visual recognition, and when I look at the line numbers referenced in the error, it is following the logic of when the visual recognition credentials are missing. But the error says "tone_analyzer" and not "visual_recognition", which is weird.

My 1st guess is that "tone_analyzer" was just a typo and it really should have said "visual_recognition", in which case, please check your `config.js` file to make sure you have the visual recognition credentials in place.

My 2nd guess is that you might be using an older version of tjbotlib (which would also explain the typo). You can do an `rm -rf node_modules && npm install`` to install the latest.

Let me know how it goes.

ryanomor, 2019-03-26:
So I would need to get "visual_recognition" credentials even if my TJBot doesn't use a camera?

ryanomor, 2019-03-26:
Thanks @jweisz. It works now after I added the "visual_recognition" credentials to my `config.js` file. I figured I wouldn't need those credentials if my TJBot doesn't use a camera. Guess that's not the case.

closed 2019-03-26 by ryanomor