ibmtjbot/tjbot #136: Apps.twitter.com has been sunset

ryanomor, 2019-03-26:
I was reading through the `sentiment_analysis` recipe and noticed that the link to get the consumer key, consumer secret, access token key, and access token secret has been sunset by Twitter.
How should we get these keys/secrets now?

maryamashoori, 2019-03-26:
Did you try https://developer.twitter.com/en/apps?

ryanomor, 2019-03-26:
I have, but not sure how to go about getting the keys/secrets from there.

jweisz, 2019-03-26:
If you figure this out, I'd appreciate a pull request to update the recipe for Twitter's new API. ;)

lpoon22, 2019-11-01:
I'm stuck here too. Was this ever figured out? @maryamashoori

jweisz, 2020-02-03:
Not sure if you figured this out already, but there is a "Keys and Tokens" section when viewing the app details. Closing this issue, but feel free to re-open if you still have questions.

closed 2020-02-03 by jweisz