ibmtjbot/tjbot #137: TJBot freezes after Sudo Node Conversation.js, followed by huge error message

zacaintmyname, 2019-04-09:
Hi team,

My credentials are updated, my speaker plays music as well. But now I just cant seem to get the last command to work! Experiencing the same issue for conversation.js and stt.js. 

pi@DoctorBroccoliJunior:~/tjbot/recipes/conversation $ sudo node conversation.jsverbose: TJBot initializing microphone
verbose: TJBot initializing LED
verbose: TJBot initializing servo motor on PIN 7
verbose: TJBot initializing assistant service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
verbose: TJBot initializing visual_recognition service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.5.1
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, what can you do?"
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone
error: the speech_to_text service returned an error. message=Session timed out., stack=Error: Session timed out.
    at emitError (/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:247:23)
    at W3CWebSocket.socket.onmessage (/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:272:17)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at WebSocketConnection.emit (events.js:180:13)
    at WebSocketConnection.processFrame (/home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:554:26)
    at /home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketConnection.js:323:40
    at process._tickCallback (internal/process/next_tick.js:176:11), type=message, isTrusted=false, _yaeti=true, data={
   "error": "Session timed out."
}, , addEventListener=function _addEventListener(type, newListener) {
	var
		listenersType,
		i, listener;

	if (!type || !newListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		this._listeners[type] = listenersType = [];
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === newListener) {
			return;
		}
	}

	listenersType.push(newListener);
}, removeEventListener=function _removeEventListener(type, oldListener) {
	var
		listenersType,
		i, listener;

	if (!type || !oldListener) {
		return;
	}

	listenersType = this._listeners[type];
	if (listenersType === undefined) {
		return;
	}

	for (i = 0; !!(listener = listenersType[i]); i++) {
		if (listener === oldListener) {
			listenersType.splice(i, 1);
			break;
		}
	}

[... 448 lines cut ...]
                    /**
                     * Object with interim or final results, possibly including confidence scores, alternatives, and word timing.
                     * @event RecognizeStream#data
                     * @param {Object} data
                     */
                    self.push(data);
                }
                else if (Array.isArray(data.results)) {
                    data.results.forEach(function (result) {
                        if (result.final && result.alternatives) {
                            /**
                             * Finalized text
                             * @event RecognizeStream#data
                             * @param {String} transcript
                             */
                            self.push(result.alternatives[0].transcript, 'utf8');
                        }
                    });
                }
            }
        }, cancelable=true, stopImmediatePropagation=function () {
			stopImmediatePropagation = true;
		}
verbose: TJBot initializing microphone
^Z2019-04-09 12:14:02 sigHandler: Unhandled signal 20, terminating

jweisz, 2019-04-09:
> error: the speech_to_text service returned an error. message=Session timed out., stack=Error: Session timed out.

Looks like the connection with Watson Speech to Text timed out. This can happen when the microphone is silent for long periods of time.

Try adding `inactivityTimeout: -1` to your tjbot config in the `config.js` file:

```
exports.tjConfig = {
    log: {
        level: 'verbose'
    },
    listen: {
        inactivityTimeout: -1
    }
};
```

Technical details here:
https://cloud.ibm.com/docs/services/speech-to-text?topic=speech-to-text-input#timeouts

andycitron, 2019-04-20:
I reported this issue last year.  I'm still having it when speaking long messages.  Setting the listen inactivityTimeout to -1 did not help.  In particular it is fatal when the exception occurs while tjbot is trying to speak.  Here are some relevant messages from my debug log:
info: TJBot heard: tell me a joke about computers 
callback from tj.listen 
... my code then starts telling a very long joke, not one of the canned jokes.  
verbose: TJBot speaking with voice en-US_MichaelV2Voice
debug: wrote audio stream to temp file /tmp/tjbot119320-2555-1edluiu.5dh0ejyvi
verbose: TJBot speaking: 
.... followed by...

debug: listening paused
debug: microphone paused
debug: Playing audio with parameters:  filename=/tmp/tjbot119320-2555-1edluiu.5dh0ejyvi, gain=100, debug=true, player=aplay, device=plughw:0,0
========= { filename: '/tmp/tjbot119320-2555-1edluiu.5dh0ejyvi',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:0,0' }
error: the speech_to_text service returned an error. message=Session timed out., stack=Error: Session timed out.
    at emitError (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:232:23)
    at W3CWebSocket.socket.onmessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:257:17)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at emitOne (events.js:96:13)
    at WebSocketConnection.emit (events.js:188:7)
.... and a long stack trace.

Why is tjbot listening at all at that point?  1) its already processing a 'return' from listening 2) its speaking and listening is paused and the mic is paused.

With all that said, if it would recover from the exception and start listening again after speaking, all would be well. After the  stack trace this is in the log:
debug: listening stopped
verbose: TJBot initializing microphone
debug: microphone deviceID set plughw:1,0
WARNING: createRecognizeStream() was renamed to recognizeUsingWebSocket(). Support for createRecognizeStream() will be removed in the next major release
debug: microphone started
debug: audio playback finished
debug: listening resumed
 
However, tjbot no longer responds to verbal commands.  It does not appear to be listening.

I'm not sure how to debug further.

andycitron, 2019-04-20:
Actually, I see now that my listen inactivity timeout setting didn't get picked up properly.  But now I did set it correctly (I think).  But it made things worse.  Upon the first attempt to listen I get this:  
verbose: TJBot initializing microphone
WARNING: createRecognizeStream() was renamed to recognizeUsingWebSocket(). Support for createRecognizeStream() will be removed in the next major release
debug: microphone started
error: the speech_to_text service returned an error. message=WebSocket connection error, stack=WebSocket connection error: WebSocket connection error
    at W3CWebSocket.socket.onerror (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:197:23)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onConnectFailed (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:217:14)
    at WebSocketClient.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:59:25)
    at emitOne (events.js:96:13)
    at WebSocketClient.emit (events.js:188:7)
    at WebSocketClient.failHandshake (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketClient.js:326:10)
    at ClientRequest.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/WebSocketClient.js:265:18)
    at emitOne (events.js:96:13)
    at ClientRequest.emit (events.js:188:7)
    at HTTPParser.parserOnIncomingClient [as onIncoming] (_http_client.js:478:21)
    at HTTPParser.parserOnHeadersComplete (_http_common.js:99:23)
    at TLSSocket.socketOnData (_http_client.js:367:20)
    at emitOne (events.js:96:13)
    at TLSSocket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18), name=WebSocket connection error, type=error, isTrusted=false, _yaeti=true, , addEventListener=function _addEventListener(type, newListener) {
	var
		listenersType,
....
If I remove this:  listen: {      inactivityTimeout: -1 } the connect works again.  So that time out value must be causing that connect exception.

VidhyaSiva, 2019-05-06:
I have getting the same error, changed the STT, TTS and conversation credentials to use password and username in config file, tried it vis swift playground and also Raspi command line. 
wondering what changed since Feb as this was working initially!!

zacaintmyname, 2019-05-06:
Hi, same for me! The service was working like clockwork in Feb for me as well!

andycitron, 2019-05-07:
Perhaps this will help.  I've been working around this problem by using pi native eSpeak to speak long passages.  With that, I still get the exception, but TJBot continues listening successfully after the exception.

The difference I see in the logs is when using eSpeak, this shows up in the log BEFORE the exception:
debug: listening paused
debug: microphone paused
debug: listening resumed

When using Watson TTS, the 'debug listening resumed' shows up after the exception and the TJBot no longer response to voice input.

It seems that resuming listening before the exception is a key timing difference that relates to the problem.

I hope this provides a clue.

jweisz, 2020-02-03:
Closing due to inactivity.

closed 2020-02-03 by jweisz

andycitron, 2020-02-03:
FWIW, I suspect this problem still exists.  I stopped using Watson text to speech for long phrases.  I found a free service that provided the same function but didn't exhibit the hang.