ibmtjbot/tjbot #139: Workspace ID Problem?

imothyt, 2019-05-03:
The first time I ran the program it seemed to work just fine, though it wouldn't output to my speakers. So, I went back and reconfigured the speakers and now every time I run the chatbot I get an error about missing the workspace ID.

verbose: TJBot initializing microphone
info: TJBot heard: Watson tell me a joke 
error: the assistant service returned an error. message=Missing required parameters: workspace_id, stack=Error: Missing required parameters: workspace_id
    at Object.getMissingParams (/home/pi/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/helper.js:94:11)
    at AssistantV1.message (/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/assistant/v1.js:95:50)
    at TJBot.converse (/home/pi/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:719:21)
    at /home/pi/tjbot/recipes/conversation/conversation.js:53:12
    at RecognizeStream.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:798:13)
    at RecognizeStream.emit (events.js:180:13)
    at addChunk (_stream_readable.js:274:12)
    at readableAddChunk (_stream_readable.js:261:11)
    at RecognizeStream.Readable.push (_stream_readable.js:218:10)
    at /home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:306:34
    at Array.forEach (<anonymous>)
    at W3CWebSocket.socket.onmessage (/home/pi/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:299:34)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at WebSocketConnection.emit (events.js:180:13)

I went into the Watson Assistant and created a sample customer care skill but now it crashes in a different way. Isn't this supposed to be able to answer a set of three or four questions (tell me a joke, etc.) straight out of the box?

25Natal, 2019-09-18:
Were you able to figure this out?

jweisz, 2020-02-03:
> Missing required parameters: workspace_id

This error means that the `workspace_id` wasn't provided in the recipe's config file. Did you import the same conversation into Watson Assistant and copy out the workspace ID? The UI for Assistant changed since we wrote the instructions for that recipe, but (I believe) the recipe should still work as written.

Closing this issue, but please re-open if you still have problems.

closed 2020-02-03 by jweisz