ibmtjbot/tjbot #143: npm install issues

mframos3, 2019-05-11:
Hi,
I am having issues with node when running npm install. Also I have trouble running sudo node on  the .js files of the recipes.

At first when using node versions 9 and above I encounter the following problem on every recipe:
```
npm WARN deprecated watson-developer-cloud@3.18.4: DeprecationWarning: watson-developer-cloud moved to ibm-watson. To get updates, use the new package.
> pigpio@0.4.0 install /home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/pigpio
> node-gyp rebuild

make: Entering directory '/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
In file included from ../../nan/nan.h:190:0,
                 from ../src/pigpio.cc:3:
../../nan/nan_maybe_43_inl.h: In function ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’:
../../nan/nan_maybe_43_inl.h:88:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
   return obj->ForceSet(GetCurrentContext(), key, value, attribs);
               ^~~~~~~~
In file included from ../src/pigpio.cc:3:0:
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:817:60: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, func, argc, argv);
                                                            ^
In file included from ../../nan/nan.h:47:0,
                 from ../src/pigpio.cc:3:
/home/pi/.node-gyp/10.15.3/include/node/node.h:177:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
... (cut due to extension)

/home/pi/.node-gyp/10.15.3/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/pigpio.cc:316:45: warning: ‘uint32_t v8::Value::Uint32Value() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   unsigned frequency = info[1]->Uint32Value();
                                             ^
In file included from /home/pi/.node-gyp/10.15.3/include/node/v8.h:26:0,
                 from /home/pi/.node-gyp/10.15.3/include/node/node.h:63,
                 from ../../nan/nan.h:47,
                 from ../src/pigpio.cc:3:
/home/pi/.node-gyp/10.15.3/include/node/v8.h:2477:47: note: declared here
   V8_DEPRECATED("Use maybe version", uint32_t Uint32Value() const);
                                               ^
... (cut due to extension. Several V8_DEPRECATED messages were shown)

rpi_ws281x.target.mk:102: recipe for target 'Release/obj.target/rpi_ws281x/src/rpi-ws281x.o' failed
make: *** [Release/obj.target/rpi_ws281x/src/rpi-ws281x.o] Error 1
make: Leaving directory '/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/rpi-ws281x-native/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.14.98-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/rpi-ws281x-native
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 

> sleep@5.2.4 install /home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/sleep
> node-gyp rebuild

make: Entering directory '/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/sleep/build'
  CXX(target) Release/obj.target/node_sleep/sleep.o
  SOLINK_MODULE(target) Release/obj.target/node_sleep.node
  COPY Release/node_sleep.node
make: Leaving directory '/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/sleep/build'

> websocket@1.0.28 install /home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/websocket
> (node-gyp rebuild 2> builderror.log) || (exit 0)

make: Entering directory '/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/websocket/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  COPY Release/bufferutil.node
  CXX(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  COPY Release/validation.node
make: Leaving directory '/home/pi/Desktop/tjbot/recipes/speech_to_text/node_modules/websocket/build'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: pigpio@0.4.0 (node_modules/pigpio):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: pigpio@0.4.0 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: rpi-ws281x-native@0.8.2 (node_modules/rpi-ws281x-native):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: rpi-ws281x-native@0.8.2 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

added 123 packages from 174 contributors and audited 298 packages in 56.197s
found 0 vulnerabilities
```
Then I tried using node version 8. I ran `npm install` with similar V8_DEPRECATED messages. I also did run `npm install ibm-watson `due to watson-developer-cloud being deprecated. After that I tried running `node stt.js`, `node conversation.js`. `node stt.js `somehow works when not using `sudo`, but `node conversation.js` requires root-privileges as it is stated on the following message: 

```
verbose: TJBot initializing microphone
verbose: TJBot initializing LED
[rpi-ws281x-native] This module requires being run with root-privileges. A non-functional stub of the interface will be returned.
verbose: TJBot initializing servo motor on PIN 7
2019-01-31 14:09:46 initCheckPermitted:
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program. |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

/home/pi/Desktop/tjbot/recipes/conversation/node_modules/pigpio/pigpio.js:11
pigpio.gpioInitialise();
```
Now by writing sudo before the command I get:
```
return process.dlopen(module, path.toNamespacedPath(filename));
^

Error: /home/pi/Desktop/tjbot/recipes/conversation/node_modules/sleep/build/Release/node_sleep.node: file too short
at Object.Module._extensions..node (internal/modules/cjs/loader.js:683:18)
at Module.load (internal/modules/cjs/loader.js:566:32)
at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
at Function.Module._load (internal/modules/cjs/loader.js:498:3)
at Module.require (internal/modules/cjs/loader.js:598:17)
at require (internal/modules/cjs/helpers.js:11:18)
at Object. (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/sleep/index.js:1:75)
at Module._compile (internal/modules/cjs/loader.js:654:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
at Module.load (internal/modules/cjs/loader.js:566:32)
at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
at Function.Module._load (internal/modules/cjs/loader.js:498:3)
at Module.require (internal/modules/cjs/loader.js:598:17)
at require (internal/modules/cjs/helpers.js:11:18)
at Object. (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:24:15)
at Module._compile (internal/modules/cjs/loader.js:654:30)
```
I have in mind that lots of issues are related to this problem, but I have tried many of the solutions people shared but none of them really worked for me.

jweisz, 2019-05-13:
Thanks for pointing out this issue. I'm not exactly sure what's going wrong here.. it looks like its failing when building the native libraries used by pigpio. I'm assuming you've done `sudo apt-get install pigpio`?

The update of the Watson SDKs from `watson-developer-cloud` to `ibm-watson` is new, but I don't think that's the reason you're getting this error.

I'll try to replicate this on my own TJBot when I get some more time. Stay tuned.

jweisz, 2019-05-15:
Alright.. tried this on my own Pi starting from a clean code checkout, using Node v9.11.2 and it works. I did get the same warnings during the `node-gyp` build of `pigpio`, but those aren't showstoppers. But I didn't get this:

```
Error: /home/pi/Desktop/tjbot/recipes/conversation/node_modules/sleep/build/Release/node_sleep.node: file too short
```

I suspect there might have been a build problem for the `sleep` module, and my recommendation would be to delete the `node_modules` directory and do another `npm install`. E.g.

```
rm -rf node_modules
npm install
```

Also, the use of `sudo` on the recipes is because root access is needed to control the LED. Both the Conversation and STT recipes use the LED, so both should be run with `sudo`.

Let me know how it goes.

stonebraker, 2019-07-25:
Hi @jweisz and @mframos3, I am experiencing the exact same thing. Were either of you able to resolve this issue? I started this summer project with my nephew and would love to complete it. :-)

If you weren't able to resolve it, may I ask which version of Raspbian you are running? I am currently running Buster. Are there other libraries I should install?

I installed pigpio with apt-get. I used node v9.11.2 with npm 5.6.0 installed via nvm. Deleted the ~/.node-gyp folder as recommended by others having add-on build issues. Removed tjbot/bootstrap/tests/node_modules directory. npm install. sudo node test.led.js.

During the installation step and the execution step, I received the exact same output as @mframos3 pasted above.

Thank you very much for your assistance!

stonebraker, 2019-07-25:
Solution seemed to be this...

First I switched to Node 10.16.0 because I was receiving errors regarding building modules with different node versions and 10.16.0 was the recommended version.

**For the pigpio issue**
`sudo apt-get install pigpio`

**For the rpi-ws281x-native issue:**
I installed rpi-ws281x-native from source. 
```git clone --recursive https://github.com/beyondscreen/node-rpi-ws281x-native.git
cd rpi-ws281x-native
npm install
node-gyp rebuild
```

I then copy pasted the resulting files into an rpi-ws281x-native directory I created in the node_modules directory.

After taking these steps, the tests/test.led.js script worked and the speech_to_text/stt.js script worked. Hope this helps anybody who might run into the same issues.

closed 2020-02-03 by jweisz