ibmtjbot/tjbot #144: Error speech to Text npm install issue

jaha0308, 2019-05-14:
pi@raspberrypi:~/tjbot/recipes/speech_to_text $ sudo npm instll
sudo np(node:25395) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.

Usage: npm <command>

where <command> is one of:
    add-user, adduser, apihelp, author, bin, bugs, c, cache,
    completion, config, ddp, dedupe, deprecate, docs, edit,
    explore, faq, find, find-dupes, get, help, help-search,
    home, i, info, init, install, isntall, issues, la, link,
    list, ll, ln, login, ls, outdated, owner, pack, prefix,
    prune, publish, r, rb, rebuild, remove, repo, restart, rm,
    root, run-script, s, se, search, set, show, shrinkwrap,
    star, stars, start, stop, submodule, t, tag, test, tst, un,
    uninstall, unlink, unpublish, unstar, up, update, v,
    version, view, whoami

npm <cmd> -h     quick help on <cmd>
npm -l           display full usage info
npm faq          commonly asked questions
npm help <term>  search for help on <term>
npm help npm     involved overview

Specify configs in the ini-formatted file:
    /root/.npmrc
or on the command line via: npm <command> --key value
Config info can be viewed via: npm help config

npm@1.4.21 /usr/share/npm
pi@raspberrypi:~/tjbot/recipes/speech_to_text $ sudo npm install
(node:25412) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN deprecated watson-developer-cloud@3.18.4: DeprecationWarning: watson-developer-cloud moved to ibm-watson. To get updates, use the new package.
 
> sleep@5.2.4 install /home/pi/tjbot/recipes/speech_to_text/node_modules/tjbot/node_modules/sleep
> node-gyp rebuild

\
Traceback (most recent call last):
  File "/usr/share/node-gyp/gyp/gyp_main.py", line 9, in <module>
    load_entry_point('gyp==0.1', 'console_scripts', 'gyp')()
  File "/usr/lib/python2.7/dist-packages/gyp/__init__.py", line 545, in script_main
    return main(sys.argv[1:])
  File "/usr/lib/python2.7/dist-packages/gyp/__init__.py", line 538, in main
    return gyp_main(args)
  File "/usr/lib/python2.7/dist-packages/gyp/__init__.py", line 502, in gyp_main
    'cwd': os.getcwd(),
OSError: [Errno 2] No such file or directory
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:304:16)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.14.98-v7+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/tjbot/recipes/speech_to_text/node_modules/tjbot/node_modules/sleep
gyp ERR! node -v v8.11.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>
npm ERR! System Linux 4.14.98-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/tjbot/recipes/speech_to_text
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 

> pigpio@0.4.0 install /home/pi/tjbot/recipes/speech_to_text/node_modules/tjbot/node_modules/pigpio
> node-gyp rebuild

I do not know what caused this problem.

jweisz, 2019-05-14:
Hm, I'm a little stumped too. Maybe try wiping out the `node_modules` folder and try again? I feel like I've run into this `gyp` error before and I think this is what I did to fix it, but it was a long time ago so my memory is a bit foggy.

```
rm -rf node_modules
npm install
```

Let me know how it goes.

jweisz, 2020-02-03:
Closing due to inactivity.

closed 2020-02-03 by jweisz