ibmtjbot/tjbot #146: tjbot-daemon.js will not start

tagoxart, 2019-05-30:
I want to connect my TJBot "water" to the iPad (playgrounds). But the tjbot-daemon.js will not start. I have entered the apikey in the line Watson Tone Analyzer in the config.js file (even if "password" and "username" is asked). As with the other Watson services: Watson Assistant, Watson Language Translator, Watson Speech to Text, Watson Text to Speech and Watson Visual Recognition. Why does not it work? What is the problem?

pi@water:~/Desktop/tjbot-daemon $ sudo node tjbot-daemon.js
verbose: TJBot initializing LED
verbose: TJBot initializing microphone
verbose: TJBot initializing servo motor on PIN 7
verbose: TJBot initializing assistant service
verbose: TJBot initializing language_translator service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
verbose: TJBot initializing tone_analyzer service
verbose: TJBot initializing visual_recognition service
/home/pi/Desktop/tjbot-daemon/node_modules/tjbot/lib/tjbot.js:542
            throw new Error(
            ^

Error: Missing authentication credentials for tone_analyzer service: username/password or apikey are required.
    at TJBot._createServiceAPI (/home/pi/Desktop/tjbot-daemon/node_modules/tjbot/lib/tjbot.js:542:19)
    at new TJBot (/home/pi/Desktop/tjbot-daemon/node_modules/tjbot/lib/tjbot.js:116:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot-daemon/tjbot-daemon.js:39:10)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:695:10)
    at startup (internal/bootstrap/node.js:201:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:516:3)
pi@water:~/Desktop/tjbot-daemon $

jweisz, 2019-06-03:
The current version of tjbotlib (1.5.1) lists the wrong service when the credential for `visual_recognition` is missing (it says `tone_analyzer`). I've fixed this in the dev branch but apparently never pushed it into master.. will do that after I post this. That said, take a look at your credentials for `visual_recognition`, maybe there is a typo? (e.g. it should be `apikey` and not `api_key`).

Once I push out the new tjbotlib (which will be 1.5.2), you can also try reinstalling to see if the error message changes:

```
rm -rf node_modules/tjbot
npm install
```

tagoxart, 2019-06-15:
Thanks for the answer. That worked. But now comes another error message. What is wrong now?

pi@water:~ $ sudo node /home/pi/Desktop/tjbot-daemon/tjbot-daemon.js
verbose: TJBot initializing Camera
verbose: TJBot initializing LED
verbose: TJBot initializing microphone
verbose: TJBot initializing servo motor on PIN 7
verbose: TJBot initializing assistant service
verbose: TJBot initializing language_translator service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
verbose: TJBot initializing tone_analyzer service
verbose: TJBot initializing visual_recognition service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.5.1
verbose: BLE state change: poweredOn
verbose: Advertising on BLE as: water
verbose: Advertising started
verbose: Creating photo directory (if needed) at /tmp/tjbot-photo/
verbose: Starting web service for /tmp/tjbot-photo/
error: error spawning http-server process
/home/pi/Desktop/tjbot-daemon/service-command/characteristic-request.js:51
        throw err;
        ^

Error: spawn node_modules/http-server/bin/http-server ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:201:19)
    at onErrorNT (internal/child_process.js:379:16)
    at process._tickCallback (internal/process/next_tick.js:178:19)

jweisz, 2020-02-03:
Apologies for taking so long to respond. It looks like there might have been a problem with the `http-server` module. The ENOENT error means that the binary it's trying to launch (`bin/http-server`) is missing.

Try reinstalling it:

```
rm -rf node_modules
npm install
```

See this SO post for more info: https://stackoverflow.com/questions/27688804/how-do-i-debug-error-spawn-enoent-on-node-js

Closing this issue, but please re-open if you still have trouble.

closed 2020-02-03 by jweisz