ibmtjbot/tjbot #148: Error: pigpio error -1 in gpioInitialise

GRuuuuu, 2019-10-11:
Hello,

> Environment
> node : v8.11.3
> npm : v5.6.0

The problem is when I run the code that using servo It makes an error below.
~~~
pi@raspberrypi:~/tjbot/recipes/conversation $ sudo node 02arm.js
verbose: TJBot initializing servo motor on PIN 7
2019-10-11 18:56:53 initInitialise: Can't lock /var/run/pigpio.pid
/home/pi/tjbot/recipes/conversation/tjbotlib/node_modules/pigpio/pigpio.js:11
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/pi/tjbot/recipes/conversation/tjbotlib/node_modules/pigpio/pigpio.js:11:12)
    at new Gpio (/home/pi/tjbot/recipes/conversation/tjbotlib/node_modules/pigpio/pigpio.js:25:3)
    at TJBot._setupServo (/home/pi/tjbot/recipes/conversation/tjbotlib/lib/tjbot.js:361:19)
    at TJBot.<anonymous> (/home/pi/tjbot/recipes/conversation/tjbotlib/lib/tjbot.js:74:22)
    at Array.forEach (<anonymous>)
    at new TJBot (/home/pi/tjbot/recipes/conversation/tjbotlib/lib/tjbot.js:59:14)
    at Object.<anonymous> (/home/pi/tjbot/recipes/conversation/02arm.js:22:10)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3
~~~

Through the google search, I can get some of the solutions. 
one of solutions like this : 
~~~
pi@raspberrypi:~/tjbot/recipes/conversation $ cat /var/run/pigpio.pid
313
pi@raspberrypi:~/tjbot/recipes/conversation $ sudo kill -9 313
pi@raspberrypi:~/tjbot/recipes/conversation $ rm /var/run/pigpio.pid
~~~

after that, I got another error.
~~~
pi@raspberrypi:~/tjbot/recipes/conversation $ sudo node 02arm.js
verbose: TJBot initializing servo motor on PIN 7
2019-10-11 18:57:28 initInitialise: bind to port 8888 failed (Address already in use)
/home/pi/tjbot/recipes/conversation/tjbotlib/node_modules/pigpio/pigpio.js:11
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/pi/tjbot/recipes/conversation/tjbotlib/node_modules/pigpio/pigpio.js:11:12)
    at new Gpio (/home/pi/tjbot/recipes/conversation/tjbotlib/node_modules/pigpio/pigpio.js:25:3)
    at TJBot._setupServo (/home/pi/tjbot/recipes/conversation/tjbotlib/lib/tjbot.js:361:19)
    at TJBot.<anonymous> (/home/pi/tjbot/recipes/conversation/tjbotlib/lib/tjbot.js:74:22)
    at Array.forEach (<anonymous>)
    at new TJBot (/home/pi/tjbot/recipes/conversation/tjbotlib/lib/tjbot.js:59:14)
    at Object.<anonymous> (/home/pi/tjbot/recipes/conversation/02arm.js:22:10)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3
~~~

+ I change hardwares (board, servo, SDcard..) but same result.
++ And, I also test hardwares in Node-red. Strangely, It works well. so The hardwares are not problems.

jweisz, 2019-10-11:
Hm, this looks like a `pigpio` issue, not a tjbot one.

Are you running on Raspian Buster? I haven't done any testing on that. It's possible some of our dependencies aren't working as expected in the new OS.

I found these from googling. There might be another service running that uses port 8888, and you should be able to hack `pigpio` to change the port to something else.

https://github.com/fivdi/pigpio/issues/7
https://github.com/fivdi/pigpio/issues/22

Let me know how it goes.

GRuuuuu, 2019-10-14:
Hello, 

Thank you so much for your help.
I figured out this problem.

To avoid crashing port, I used `configureSocketPort` function and set the default port 8888 to 8889
> to use `configureSocketPort`, You must upgrade `pigpio` to 0.6.0 or higher.  

So I changed the code tjbotlib/lib/tjbot.js
~~~js
TJBot.prototype._setupServo = function(pin) {
    var pigpio=require('pigpio');
    pigpio.configureSocketPort(8889);
    var Gpio=pigpio.Gpio;
    ....
~~~

And It worked well for me.

By the way, I think all the problem that I have is caused by old dependency. (specially pigpio, I used v0.4.0)
So In the recent version of code, It may not happen.

+) Additionally In my case, The instance holding port 8888 is Node-Red

closed 2019-10-14 by GRuuuuu

jweisz, 2019-10-14:
Awesome, I'm glad you figured it out!