ibmtjbot/tjbot #149: Pi 4 support

d1runberg, 2019-10-18:
Is Tjbot supported using the pi 4? The docs say nothing about it and want to be sure before making recommendations on hardware.

jweisz, 2019-10-21:
From a software side, I believe everything should work, but I haven't tested it myself yet.

From a hardware side, we have a community member who altered the design of the 3D print to accommodate the RPi 4. @maryamashoori has more info on the status of this.

d1runberg, 2019-10-21:
Awesome! Do we know if the cardboard laser cut files work, or do they need
to be modified in some way?

*Derek Runberg*
*Strategic Partnerships / Author / Educator*

*SparkFun Electronics, Inc.*
p: 303.284.0979 <(303)%20284-0979>  m: 971.344.2355
a: 6333 Dry Creek Pkwy
  Niwot, CO. 80503
w: sparkfun.com  e: derek.runberg@sparkfun.com <first.last@sparkfun.com>
<https://facebook.com/sparkfun>   <https://twitter.com/sparkfun>
<https://instagram.com/sparkfun> <https://youtube.com/sparkfun>

chunyianliew, 2019-11-01:
The 3D print for the Raspberry Pi 4 is almost the same as for the Raspberry Pi 3 B+, except for the jaw. The holes in the jaw for the usb and network ports needs to be mirrored.
The tjbot node-red install process almost results in a working tjbot. The only issue I ran into so far is that when the led enabled in the tjbot node configuration the deployment fails/hangs/loops...

chunyianliew, 2019-11-03:
I managed to get the led working on a Raspberry Pi 4.
The led does not initialise with the imported node_rpi_ws281x-native module. It does not report any errors, it just hangs when trying to initialising the led control.
This issue is also described [here](https://github.com/beyondscreen/node-rpi-ws281x-native/issues/99).

I managed to resolve the issue for now by replacing the imported rpi_ws281x library (of a specific tree version) in the source folder of node-rpi-ws281x-native by the [master branch](https://github.com/beyondscreen/rpi_ws281x).

Here are the commands when running node-red:

_\# cd ~/.node-red/nodes/node-red-contrib-tjbot/node_modules/rpi-ws281x-native/src
\# rm -rf rpi_ws281x/
\# git clone --recursive https://github.com/beyondscreen/rpi_ws281x.git
\# cd ~/.node-red/nodes/node-red-contrib-tjbot/node_modules/rpi-ws281x-native
\# node-gyp rebuild_

So tjbot is now fully operational on a Raspberry Pi 4.

jweisz, 2020-02-03:
Closing due to inactivity.

closed 2020-02-03 by jweisz

CaskAle, 2020-05-28:
> Is Tjbot supported using the pi 4? The docs say nothing about it and want to be sure before making recommendations on hardware.

There are a few SW issues that must be dealt with.  You can see the patches described in this doc. https://github.com/CaskAle/summit-pi-project.  Or by reading these issues re. the LED.