ibmtjbot/tjbot #151: Import the workspace-sample.json file into the Watson Assistant service

DavidMcCrorie, 2019-11-22:
Please help on how to Import the workspace-sample.json file into the Watson Assistant service .

This is part of Step 5 in the Recipe called TJBot - Build a Talking Robot
https://www.instructables.com/id/Build-a-Talking-Robot-With-Watson-and-Raspberry-Pi/
https://github.com/ibmtjbot/tjbot/tree/master/recipes/conversation

It's not obvious to me.     I think it might have been obvious in the old Watson Conversation.

jweisz, 2019-11-25:
Hi @DavidMcCrorie, thanks for pointing this out. It looks like the Watson Assistant UI was recently redesigned, so the old instructions won't work. Please try the following and let me know if they work:

1. From the Watson Assistant service in IBM Cloud, click "Launch Watson Assistant"
2. Click “Create assistant”
3. Give it a name (e.g. "tjbot") and click “Create”
4. Click “Add a dialog skill”
5. Click the “Import skill” tab
6. Click “Choose JSON file”
7. Select “workspace-sample.json”
8. Click “Import”
9. At this point, you should see the new Dialog skill you just created
10. Click the vertical ellipsis menu (three vertical dots) and select "Settings"
11. Click "API Details" in the left sidebar
12. Copy the Assistant ID and paste into the variable `exports.workspaceId` in your config.js file.
13. Copy the Api Key and paste into the `apikey` variable in your config.js file (under `exports.credentials.assistant`.

DavidMcCrorie, 2019-11-25:
Thank you.   These instructions helped me move forward.    

I performed a step 8a to click the Watson Assistant label at top left of screen.   This changed back to the main page so when I clicked the 3 vertical dots, I could then see the Settings option.

closed 2020-02-03 by jweisz