ibmtjbot/tjbot #152: when running conversation.js, having issues with assistant service

sparksparker, 2019-12-05:
TJBot is recognizing words, but when it is triggered by its name, i.e. Watson, the following error occurs. I have not yet had the chance to hear Watson respond.

error: the assistant service returned an error. message=Resource not found, stack=Error: Resource not found
    at Request._callback (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/requestwrapper.js:102:21)
    at Request.self.callback (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:185:22)
    at Request.emit (events.js:189:13)
    at Request.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:1157:10)
    at Request.emit (events.js:189:13)
    at IncomingMessage.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:1079:12)
    at Object.onceWrapper (events.js:277:13)
    at IncomingMessage.emit (events.js:194:15)
    at endReadableNT (_stream_readable.js:1103:12)
    at process._tickCallback (internal/process/next_tick.js:63:19), code=404, error=Resource not found, x-global-transaction-id=ebe14a03c58de97d0ebdf0028bccdc42

I think I'm just missing small here. 

Any suggestions?

Thanks in advance!

jweisz, 2019-12-05:
Hi @sparksparker, quick question: did you set your workspace ID in the `config.js` file? I suspect that having an empty workspace ID would trigger a "Resource not found" error, but I'm unable to confirm this right now.

If you did set the workspace ID, it might be an issue with the fact that we're still using the older `watson-developer-cloud` SDK rather than the newer `ibm-cloud` SDK. I'm in the middle of trying to migrate `tjbotlib` to use the newer SDK, but it hasn't been a smooth transition and I haven't had much time to work on it.

sparksparker, 2019-12-05:
@jweisz thanks for getting back to me

yes, i've set the workspace ID in the config.js file from 
assistant settings => assistant ID

is there a workaround i could use in the meantime?

sparksparker, 2019-12-05:
sorry, this was error on my part. thank you for your help

the issue was, i was using the **workspace ID** obtained from 
assistants => 
settings => 
api details => 
assistant ID

it should actually be

assistants => 
(whatever you named your assistant i.e. watson) => 
specific dialog - 3 dot menu => 
view API details => 
skill ID

hope this helps others as well

closed 2019-12-06 by sparksparker

jweisz, 2019-12-06:
👍 glad you figured it out!

chbarrer, 2020-03-01:
Hi!.. Have same problem:

verbose: TJBot initializing microphone
verbose: TJBot initializing LED
verbose: TJBot initializing servo motor on PIN 7
verbose: TJBot initializing Camera
verbose: TJBot initializing assistant service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
verbose: TJBot initializing visual_recognition service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.5.1
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, what can you do?"
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone
info: TJBot heard: Watson tell me a joke 
**error: the assistant service returned an error. message=Resource not found, stack=Error: Resource not found
    at Request._callback (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/requestwrapper.js:102:21)
    at Request.self.callback (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:185:22)
    at emitTwo (events.js:106:13)
    at Request.emit (events.js:191:7)
    at Request.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:1157:10)
    at emitOne (events.js:96:13)
    at Request.emit (events.js:188:7)
    at IncomingMessage.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:1079:12)
    at IncomingMessage.g (events.js:292:16)
    at emitNone (events.js:91:20)
    at IncomingMessage.emit (events.js:185:7)
    at endReadableNT (_stream_readable.js:974:12)
    at _combinedTickCallback (internal/process/next_tick.js:80:11)
    at process._tickCallback (internal/process/next_tick.js:104:9), code=404, error=Resource not found, x-global-transaction-id=da1e1863557dee3c0ca73f3f80c6e973
^C2020-03-01 15:12:12 sigHandler: Unhandled signal 2, terminating**

Tried changing workspace ID by skill ID as @sparksparker did, but still same problem.. any ideas ?..

mirjamblekkenhorst, 2020-03-01:
For me the conversation part is workingnow, although the tone analyser is not yet...

But just a thought ... have you copied the Tjbot in skills (instead of the copy into the workspace) ? And then copied the skills ID from this skill set?

[cid:E066935D-BAEB-4F5A-B740-6425018ECC6A]

Met vriendelijke groet,

Mirjam Blekkenhorst

at Request. (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:1157:10)
at IncomingMessage. (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:1079:12)

chbarrer, 2020-03-01:
Hi  @mirjamblekkenhorst, thanks for your reply. 

First time follow instructions here: 

https://github.com/ibmtjbot/tjbot/issues/151

In this we create assistant and then a dialog skill, and then import the workspace-sample.JSON file, and follow instructions.

Last try was using ID from Skill tab.