ibmtjbot/tjbot #153: When running stt.js

sgsilva, 2020-01-30:
when a running stt.js im freezing in line ( TjBOT initializing LED ) after this i cant open another terminal or anything.
check the picture.
[image]

chunyianliew, 2020-01-30:
Are you by any chance running on a Raspberry Pi 4?

sgsilva, 2020-01-30:
Yes, in Pi 4 B+.

chunyianliew, 2020-01-30:
Then please take a look at this comment: https://github.com/ibmtjbot/tjbot/issues/149#issuecomment-549189988

sgsilva, 2020-01-30:
`I managed to get the led working on a Raspberry Pi 4.
The led does not initialise with the imported node_rpi_ws281x-native module. It does not report any errors, it just hangs when trying to initialising the led control.
This issue is also described here.

I managed to resolve the issue for now by replacing the imported rpi_ws281x library (of a specific tree version) in the source folder of node-rpi-ws281x-native by the master branch.

Here are the commands when running node-red:

# cd ~/.node-red/nodes/node-red-contrib-tjbot/node_modules/rpi-ws281x-native/src
# rm -rf rpi_ws281x/
# git clone --recursive https://github.com/beyondscreen/rpi_ws281x.git
# cd ~/.node-red/nodes/node-red-contrib-tjbot/node_modules/rpi-ws281x-native
# node-gyp rebuild` 
i dont find  ~/.node-red/nodes/node-red-contrib-tjbot/node_modules/rpi-ws281x-native/src
im searching but i dont have sucess

chunyianliew, 2020-01-30:
As you are not running node-red, but probably installed tjbot using the bootstrup url on your Desktop folder, you need to replace the first part all reference to the directory _~/.node-red/nodes/node-red-contrib-tjbot/_ with the directory where the folder _node_modules_ resides.

I don't know the exact location by heart and it will take a few hours before I have access to a raspberry pi again, but it should be a subdirectory somewhere under  _/home/pi/Desktop/tjbot/_ as shown in your screenshot.

chunyianliew, 2020-01-30:
I just cloned the tjbot project and installed the dependencies on my mac, I found the location:
/home/pi/Desktop/tjbot/bootstrap/tests/

So this should work for you:

```
# cd /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native/src
# rm -rf rpi_ws281x/
# git clone --recursive https://github.com/beyondscreen/rpi_ws281x.git
# cd /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native
# node-gyp rebuild
```

sgsilva, 2020-01-30:
yes, thank you sir but i dont have this past rpi-ws281x-native/src

sgsilva, 2020-01-30:
my install log in past test.

`npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/
npm WARN deprecated watson-developer-cloud@3.18.4: DeprecationWarning: watson-developer-cloud moved to ibm-watson. To get updates, use the new package.
npm WARN deprecated @types/form-data@2.5.0: This is a stub types definition. form-data provides its own type definitions, so you do not need this installed.

> pigpio@1.3.0 install /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/pigpio
> node-gyp rebuild

gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/pigpio/build'
gyp ERR! System Linux 4.19.75-v7l+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/pigpio
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 

> rpi-ws281x-native@0.9.0 install /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native
> node-gyp rebuild

gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native/build'
gyp ERR! System Linux 4.19.75-v7l+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 

> sleep@5.2.4 install /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/sleep
> node-gyp rebuild

gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/sleep/build'
gyp ERR! System Linux 4.19.75-v7l+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/sleep
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: pigpio@1.3.0 (node_modules/pigpio):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: pigpio@1.3.0 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: rpi-ws281x-native@0.9.0 (node_modules/rpi-ws281x-native):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: rpi-ws281x-native@0.9.0 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sleep@5.2.4 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the sleep@5.2.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-01-30T15_26_09_348Z-debug.log
`

chunyianliew, 2020-01-30:
Ah, so you were missing the necessary libraries.
I suppose the easiest solution would be to downgrade the node version to 9: https://www.surrealcms.com/blog/how-to-upgrade-or-downgrade-nodejs-using-npm.html

Then re-install the tjbot dependencies again: 
`npm install ~/Desktop/tjbot/bootstrap/tests`

Use `sudo npm install ~/Desktop/tjbot/bootstrap/tests` if you run into permission issues.

And try to run the tests again when ready. If the led initialisation still hangs, try replacing the rpi_ws281x library with the commands provided earlier.

sgsilva, 2020-01-30:
thank you again friend.

`Resolving deltas: 100% (121/121), done.
pi@cain:~/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native/src $ cd /home/pi/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native
pi@cain:~/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native $ sudo node-gyp rebuild
internal/modules/cjs/loader.js:550
    throw err;
    ^

Error: Cannot find module 'graceful-fs'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:548:15)
    at Function.Module._load (internal/modules/cjs/loader.js:475:25)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/usr/share/node-gyp/lib/node-gyp.js:12:10)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
pi@cain:~/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native $ 
`

when i use sudo node-gyp rebuild

chunyianliew, 2020-01-30:
Did you run into any issues when downgrading to node version 9.11.2?

It looks like your npm is not installed properly: https://techoverflow.net/2012/10/07/how-to-fix-npm-cannot-find-module-graceful-fs-error/

sgsilva, 2020-01-30:
yes friend im using 9.11.2.

   installed : v9.11.2 (with npm 5.6.0)
pi@cain:~/Desktop/tjbot/bootstrap/tests/node_modules/rpi-ws281x-native $

chunyianliew, 2020-01-30:
Sorry, I have not run into the latest issue myself. My recommendation would be to look into similar references at github. Or wait until maybe somebody else has a suggestion in this thread.

sgsilva, 2020-01-30:
thank you friend, for all!
i can run speaker test and ok, only test,led freezing.

jweisz, 2020-02-03:
Try starting from a clean checkout of tjbot.

```
cd ~/Desktop
git clone https://github.com/ibmtjbot/tjbot.git
cd tjbot/bootstrap/tests
npm install
```

Now check out the branch of rpi-ws281x that supports RPi 4:

```
npm install rpi-ws281x-native@latest
git clone --single-branch --branch raspi4support https://github.com/jimbotel/rpi_ws281x.git
cp -r rpi_ws281x/* node_modules/rpi-ws281x-native/src/rpi_ws281x
npm build node_modules/rpi-ws281x-native
```

Now run the recipe:

```
sudo node test.led.js
```

Please note that you will need to check out the rpi-ws281x branch for *each* recipe you want to use. E.g. if you want to try the speech to text recipe, you'll need to do the same set of commands in the 2nd code block above to enable the LED to work.

melaurean, 2020-03-04:
@jweisz Thanks Justin! I started fresh and will keep you posted.

chunyianliew, 2020-03-25:
@melaurean, FYI I've executed the instructions provided by @jweisz and I successfully managed to get the led test working. See attached file for the logs.

I've used a Raspbian Buster Image (with desktop and recommended software), Version February 2020, Release date 2020-02-13, Kernel version: 4.19, Size: 2530 MB.

One comment though, when starting up for the first time, the Desktop folder used in the instructions did not exist yet because I am always using the Raspberry Pi headless.

[tjbot_issue_153_validate.txt](https://github.com/ibmtjbot/tjbot/files/4380630/tjbot_issue_153_validate.txt)

melaurean, 2020-03-25:
@chunyianliew, Congratulations! I was running  @jweisz's instructions on Pi4 with the latest Raspi Buster version. I will check my OS details. In all honesty, I gave up on LEDs and moved to other AI topics. I would love to find my issue, and I plan to revisit the subject. Thanks again for keeping us abreast.

melaurean, 2020-03-27:
@chunyianliew, if you would please let me know which RPI V4 H/W have you tested with, i.e. the 1, 2, or 4GB, and if you do not mind run a uname -m command and let me know of the result. 

FYI I tried all Raspbian images and I am inclined to believe that I am facing a H/W bord issue. While testing with the Buster image you mentioned above the Desktop dir was created. Nevertheless, I installed the TJbot code in a sandbox . Thanks for your support!

chunyianliew, 2020-03-27:
@melaurean, I am using a Raspberry Pi 4 with 1 GB memory.

`pi@raspberrypi:~ $ free -m
              total        used        free      shared  buff/cache   available
Mem:            872         212         309          48         351         554
Swap:            99           6          93
pi@raspberrypi:~ $ uname -m
armv7l
`
As the led needs to access the DMA Controller with root privileges, could it be that this is restricted when using a sandbox environment one way or another?

melaurean, 2020-03-29:
Gents, @jweisz, @chunyianliew the LED saga is ongoing for me. Professional curiosity forces me to dig further -- as time permits. I built fresh SD cards -- every 2020-02-13 Booster release--and followed @jweisz's steps listed above. The test fails as I described in this thread on the RPI V4 hardware. However, the very same SD works perfectly on RPI V3 hardware. 
The last resort I can think of is to trace the kernel. Please let me know if you know of a good raspberry kernel debugger, so I start on solid ground.

chunyianliew, 2020-03-30:
@melaurean, sorry I have no experience with kernel debuggers myself.
What I did when I ran into the tjbot initialisation issue with the RPI 4 myself was to change the log level to debug and looked into the nodejs log file.
And also tailed /var/log/kern.log, /var/log/messages and /var/log/syslog.
That's how I discovered that the code kept hanging somewhere in the rpi-ws281x module library.

melaurean, 2020-03-30:
@chunyianliew, good thought. I checked the logs, but unfortunately when the kernel dies the logs are behind the execution stack .  Maybe @jweisz could advise. Thanks a lot for all your support!

jweisz, 2020-11-13:
Closing this issue as instructions for making neopixel LEDs work on RPi4 have been included in the TROUBLESHOOTING.md guide as part of the tjbot 2.0.0 release.

closed 2020-11-13 by jweisz