ibmtjbot/tjbot #154: Error when i ask tjbot

sgsilva, 2020-02-01:
B, name of tjbot.
when i ask him i got erros.

`info: TJBot heard: B. what can you do 
error: the assistant service returned an error. message=Missing required parameters: workspace_id, stack=Error: Missing required parameters: workspace_id
    at Object.getMissingParams (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/helper.js:94:11)
    at AssistantV1.message (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/assistant/v1.js:95:50)
    at TJBot.converse (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:719:21)
    at /home/pi/Desktop/tjbot/recipes/conversation/conversation.js:53:12
    at RecognizeStream.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:798:13)
    at RecognizeStream.emit (events.js:180:13)
    at addChunk (_stream_readable.js:274:12)
    at readableAddChunk (_stream_readable.js:261:11)
    at RecognizeStream.Readable.push (_stream_readable.js:218:10)
    at /home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:306:34
    at Array.forEach (<anonymous>)
    at W3CWebSocket.socket.onmessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/lib/recognize-stream.js:299:34)
    at W3CWebSocket._dispatchEvent [as dispatchEvent] (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/yaeti/lib/EventTarget.js:107:17)
    at W3CWebSocket.onMessage (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:234:14)
    at WebSocketConnection.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/websocket/lib/W3CWebSocket.js:205:19)
    at WebSocketConnection.emit (events.js:180:13)
`

jweisz, 2020-02-03:
Hi @sgsilva , I'm not sure I see the error above. It looks like the recipe worked correctly. What was the error?

sgsilva, 2020-02-03:
i was solved, i chenge de apikey and work, thanl you friend!

closed 2020-02-03 by jweisz