ibmtjbot/tjbot #158: I managed to get the led working on a Raspberry Pi 4.

melaurean, 2020-03-03:
I managed to get the led working on a Raspberry Pi 4.
The led does not initialise with the imported node_rpi_ws281x-native module. It does not report any errors, it just hangs when trying to initialising the led control.
This issue is also described [here](https://github.com/beyondscreen/node-rpi-ws281x-native/issues/99).

I managed to resolve the issue for now by replacing the imported rpi_ws281x library (of a specific tree version) in the source folder of node-rpi-ws281x-native by the [master branch](https://github.com/beyondscreen/rpi_ws281x).

Here are the commands when running node-red:

_\# cd ~/.node-red/nodes/node-red-contrib-tjbot/node_modules/rpi-ws281x-native/src
\# rm -rf rpi_ws281x/
\# git clone --recursive https://github.com/beyondscreen/rpi_ws281x.git
\# cd ~/.node-red/nodes/node-red-contrib-tjbot/node_modules/rpi-ws281x-native
\# node-gyp rebuild_

So tjbot is now fully operational on a Raspberry Pi 4, at least mine tjbot is.

_Originally posted by @chunyianliew in https://github.com/ibmtjbot/tjbot/issues/149#issuecomment-549189988_

melaurean, 2020-03-03:
I just started TJbot High School project using Pi 4 and ran into the LED issue described by @chunyianliew. I applied the commands above to my node v10.15.2 environment with no success. The node-gyp returned OK, but the stty.js (../recipes/speech-to-text) blocks when ran with sudo privileges.
The H/W tests are looking for a 'readline-sync' module which I could not locate, so I could not run test.led.js. Any help would be highly appreciated. I can't revert back to Pi 3 since the school already has Pi 4.

jweisz, 2020-03-03:
If `readline-sync` is missing, you may need to do an `npm install`?

melaurean, 2020-03-03:
Thank you, we'll do!. Any thoughts on the first issue? Failure occurs while TJBot is initializing LED. I attempted many recommended fixes short of changing DMA settings.

chunyianliew, 2020-03-03:
@melaurean, I am organising similar workshops at work for kids of colleagues, so I can share some of my own findings I ran into myself with the Led, assuming that the node-gyp rebuild didn't throw any errors:

- Please pay attention to the data sheet of the leds, if you have sourced the Neopixel 8mm led from Adafruit, this should be fine. But when you sourced cheaper alternatives (because I need quite a lot of them, 8 leds per workshop), I got 8mm Leds with a WS2811 chipset. But the pin layout of the leds I received differed from the Neopixel from Adafruit.
- I am not certain why you are mentioning DMA settings, but one issue I ran into is that I tried to adjust the code base of the project to allow me to control the led light without root privileges. But that seems not to be possible for the time being. A colleague of mine, @ebaauw, he digged into this issue and explained the following to me: "the led controller expects a datastream of 8-bit RGB values based on a 800 Mhz clock signal. A regular computer can not sent a stable clock signal due to CPU interrupts, so a real time system is required for the timing. For the Pi, they have used a brilliant hack: they use the DMA controller to send the data stream from memory to the GPIO pin. However, this hack makes use of the same DMA channel as the on-board audio, hence the need to use a USB or Bluetooth Speaker. That's also why you will see the led starts to blink when you use the on-board audio (speak command)." You need root privileges to make use of the DMA controller.

melaurean, 2020-03-03:
@chunyianliew, many thanks for sharing your findings. I tried both NeoPixel and DIGI_KEY LEDs and so far with little success. All works fine on Pi 3, but not yet on Pi 4.  Since the failure occurs in the LED init code I researched that aspect and found a few posts mentioning the need to change DMA channel on Pi4 to 10. I started reading the code but did not jump into making the change yet...

chunyianliew, 2020-03-03:
@melaurean, another option is to try out the following suggestion from @jweisz: https://github.com/ibmtjbot/tjbot/issues/153#issuecomment-581564613

If you have some patience: I am planning to create a new golden master SD card image for the Raspberry Pi 4 within the coming 2 weeks. I use this image to clone multiple SD cards in one go for the workshops. I can share the instructions to create one and the actual image with you when I am ready. The workshop is fully focused on usage of Node-Red with the tjbot though.

melaurean, 2020-03-04:
@jweisz Is there a list of S/W component versions that I should observe for TJ? For example the documentation recommends back versions of node.js (6 & 9).

melaurean, 2020-03-06:
> (comment)

Sounds good. The LED issue seems to be related to a a misunderstanding
between the initialization routine  ws281x-native  and RPI. Do you plan to
change that locally, or work with the OS team for a more generic approach?

jweisz, 2020-11-13:
Closing this issue as the tjbot 2.0.0 release adds instructions for getting neopixel LEDs working on RPi 4 in the new TROUBLESHOOTING.md guide, as well as support for common anode LEDs.

closed 2020-11-13 by jweisz