ibmtjbot/tjbot #160: Play audio in a bluetooth speaker

pdro7, 2020-08-27:
Hi everyone, I am struggling trying to make TjBot to use a Bluetooth speaker as audio output,  right from the Graphical Desktop Interface if I play some video in  youtube I can listen it in the Bluetooth speaker, so it is working, but I can not make it work for TJBot, in the config.js file I have the folow lines, I am not sure about the speakerDeviceId, I have been reading several post and articles online to make the bluetooth speaker works.  I will appreciate any help

// set up TJBot's configuration
exports.tjConfig = {
    speakerDeviceId: 'bluez_card.FC_58_FA_C1_AD_28',
    microphoneDeviceId: 'plughw:1,0',
    log: {
        level: 'verbose'
    },
    // Changing the robot name will change the attention word
     robot: {
        name: 'Jack'
        //gender: 'female'
     },
     speak: {
        speakerDeviceId:"bluez_card.FC_58_FA_C1_AD_28"
        }
};

chunyianliew, 2020-09-07:
Have you already seen this similar issue report?: https://github.com/ibmtjbot/tjbot/issues/69.

If the steps mentioned there don't resolve you issue, please show the error message when you try to run the speaker test:

`pi@tjbot:~/Desktop/tjbot/bootstrap/tests $ node test.speaker.js `

jweisz, 2021-10-21:
Closing this issue due to inactivity.