ibmtjbot/tjbot #161: Conversation Recipes showing error that WorkspaceID is not provided

erramilli, 2020-09-06:
I have followed everything in the documentation and this onething seems to be confusing and frustrating at the same time.

As instructed I have 

1. Configured all the 3 services (STT , TexttoSpeech, Watson Assistant).

2. Imported the Workplace json into the Skill in Watson Assistant.

3. I have updated all the API Key etc.. in the the Config.Js

Ran SUDO node conversation.js

The program starts...recognizes everything I say... (basically proves my STT is working..right ?)
The moment I utter the word 'Watson'  it crashes with a message a required parameter Workspace_id is not provided

When I go to  WatsonAssistant-->skill --> API Details ..  I don't really see something with a name  WorkSpaceID .

All I see is   'Legacy Workspace URL'  ..  If not where can I find the workspaceID ?.. 

Are these two the same ?

chunyianliew, 2020-09-06:
You need to fill in the Skill ID here. I have documented it here where you can find this value: [https://aibot.liew.nl/en/preparations/ibm-cloud](https://aibot.liew.nl/en/preparations/ibm-cloud)

erramilli, 2020-09-06:
Thank you @chunyianliew . I still see the error as below, even after using the SkillID as WorkspaceId.. not exactly sure what i'm missing.

===
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone
info: TJBot heard: Watson 
error: the assistant service returned an error. message=Missing required parameters: workspace_id, stack=Error: Missing required parameters: workspace_id
    at Object.getMissingParams (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/helper.js:94:11)
    at AssistantV1.message (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/assistant/v1.js:95:50)
    at TJBot.converse (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:719:21)
    at /home/pi/Desktop/tjbot/recipes/conversation/conversation.js:53:12

Here is my how my config.js looks like

exports.workspaceId = '03axxxxx-71e2-4a54-84cc-235b20xxxxxx';

// Set this to false if your TJBot does not have a camera.
exports.hasCamera = false;

// set up TJBot's configuration
exports.tjConfig = {
    log: {
        level: 'verbose'
    },
    // Changing the robot name will change the attention word
    // robot: {
    //    name: 'tee jay bot'
    // }
};

chunyianliew, 2020-09-06:
Please check that the Watson Assistant url configured in the config.js file in the recipes/conversation directory contains the same url as shown in the API details of your IBM Cloud Watson Assistant account.

In the latest version of config.default.js file i see the following line:
`    url: 'https://gateway.watsonplatform.net/assistant/api/'`

When I login I see the following url (for location London) starting with `https://api.eu-gb.assistant.watson.cloud.ibm.com/instances`

I experienced issues with changed url end-points a few months ago when using node-red, your issue might be related: https://github.com/chunyianliew/node-red-contrib-tjbot/pull/1

weirending, 2020-09-06:
Maybe you need to set workspace_id In stead of workspaceId?

erramilli, 2020-09-06:
Still not working...  .. Based on @weirending suggestion I have added workspaceid  also

**exports.workspaceId = '03acb137-71e2-XXXX-XXXX-235b20cb3fe2';
exports.workspace_id = '03acb137-71e2-XXXX-XXXX-235b20cb3fe2';**
// Set this to false if your TJBot does not have a camera.
exports.hasCamera = false;

// set up TJBot's configuration
exports.tjConfig = {
    log: {
        level: 'verbose'
    },
};
// Create the credentials object for export
exports.credentials = {};
// Watson Assistant
// https://www.ibm.com/watson/services/conversation/
exports.credentials.assistant = {
      **apikey: 'yZRLg-3svkzpW-XXXXXXPDYUWXh8T0HJEEdyORXRXA58',
      url: 'https://api.us-south.assistant.watson.cloud.ibm.com/instances/a8e8e354-d8de-XXXX-XXXX-bb88b8a53323'**
};

erramilli, 2020-09-06:
Here is the message I see...... still

info: TJBot heard: Watson 
error: the assistant service returned an error. message=Missing required parameters: workspace_id, stack=Error: Missing required parameters: workspace_id
    at Object.getMissingParams (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/helper.js:94:11)
    at AssistantV1.message (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/watson-developer-cloud/assistant/v1.js:95:50)
    at TJBot.converse (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js:719:21)

chunyianliew, 2020-09-06:
Can you please share the full command line you use to start the converse recipe? Please also include the output of the pwd command to determine from which directory the script is being called. Please also share the absolute path to the config.js file.

erramilli, 2020-09-06:
@chunyianliew 

here is the command I'm using..
pi@Pbot:~/Desktop/tjbot/recipes/conversation $ **sudo node conversation.js**

config.js is in   **/home/pi/Desktop/tjbot/recipes/conversation**

output of pwd  :    **/home/pi/Desktop/tjbot/recipes/conversation**

chunyianliew, 2020-09-06:
Your setup looks good so I took a look at the documentation: https://cloud.ibm.com/apidocs/assistant-icp
There it states the following:
```
The service endpoint for Watson Assistant v1 is always the following base URL:

`https://{icp_cluster_host}{:port}/assistant/api`
```

So can you please change the url for Watson in the config.js file to `https://api.us-south.assistant.watson.cloud.ibm.com/assistant/api` ?

erramilli, 2020-09-06:
@chunyianliew       you are suggesting to change like this ??  ... didn't work.  Also, Thank you so much for your quick responses.

exports.credentials = {};
// Watson Assistant
// https://www.ibm.com/watson/services/conversation/
exports.credentials.assistant = {
      apikey: 'yZRLg-3svkzpW-RBVhzfPDYUWXh8T0HJEEdyORXRXA58',
     // url: 'https://api.us-south.assistant.watson.cloud.ibm.com/instances/a8e8e354-d8de-4327-929a-bb88b8a5yyyy'//,
      **url: 'https://api.us-south.assistant.watson.cloud.ibm.com/assistant/api/v1/a8e8e354-d8de-4327-929a-bb88b8a5yyyy**
      
};

chunyianliew, 2020-09-06:
No, please use:

url: 'https://api.us-south.assistant.watson.cloud.ibm.com/assistant/api/'

I suspect that the full url shown in the IBM cloud administration area is the V2 url including the reference to the assistant as mentioned on the documentation. That additional part was not used in V1.

If this doesn't work out I will hookup a tjbot tomorrow and try it myself tomorrow and let you know if I can it get it to work with the sample conversation recipe. I am located in the Netherlands and it is getting late...

erramilli, 2020-09-06:
@chunyianliew  :  still same issue..

I'm suspecting something is stopping it from loading the workspaceid from the config.js ... however, as the speechtotext portion is working .. I must believe it is infact able to locate the config.js file and read the contents..

chunyianliew, 2020-09-06:
That's also why I requested to double check the location of the config.js file. Sometimes the error messages returned by the IBM cloud js node modules do not correctly represent the actual error, that is speaking from my own personal experience with the tjbot project. I will try to see if I run into the same issue tomorrow.

chunyianliew, 2020-09-07:
@erramilli I have installed the tjbot recipes and I managed to get it to work. I did ran into two known issues. One because I am using a Raspberry Pi 4 and the other one because of outdated tjbot node module code:

- https://github.com/ibmtjbot/tjbot/issues/149#issuecomment-549189988

- https://github.com/chunyianliew/tjbotlib/commit/e57e52b7fdf21bef97eee9a45773b359804285ae

My suggestion would be to delete your current conversation directory and start over again and share the output of the install process:
```
$ cd tjbot/recipes/conversation
$ npm install 
```
With regard to the url's used in the config.js file, here are the URL's that worked for me:
Watson Assistant:     url: 'https://api.eu-gb.assistant.watson.cloud.ibm.com/instances/092fe60a-95af-4f54-9afc-766887972fc5' (Location: London, copied full URL from the IBM Cloud Service credentials overview) 
Speech to Text:     url: 'https://api.eu-gb.speech-to-text.watson.cloud.ibm.com'  (Location: London, copied only the first part of the URL from the IBM Cloud Service credentials overview, the full URL might work also, haven't tested it myself)
Text to Speech:     url: 'https://api.eu-gb.text-to-speech.watson.cloud.ibm.com' (Location: London, copied only the first part of the URL from the IBM Cloud Service credentials overview, the full URL might work also, haven't tested it myself)
Watson Visual Recognition:     url: 'https://api.eu-de.visual-recognition.watson.cloud.ibm.com' (Location: Frankfurt, copied only the first part of the URL from the IBM Cloud Service credentials overview, , the full URL might work also, haven't tested it myself)

Please also change the debug level in config.js from verbose to debug and share the startup log with us for further investigation:

This is is how my startup log looks like:
```
pi@tjbot:~/Desktop/tjbot/recipes/conversation $ sudo node conversation.js 
verbose: TJBot initializing microphone
verbose: TJBot initializing LED
verbose: TJBot initializing servo motor on PIN 7
verbose: TJBot initializing Camera
verbose: TJBot initializing assistant service
verbose: TJBot initializing speech_to_text service
verbose: TJBot initializing text_to_speech service
verbose: TJBot initializing visual_recognition service
info: Hello from TJBot! My name is Watson.
verbose: TJBot library version v1.5.1
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, what can you do?"
You can also say, "Watson, tell me a joke!"
verbose: TJBot initializing microphone
debug: microphone started
(node:1512) Warning: N-API is an experimental feature and could change at any time.
info: TJBot heard: Watson what do you see 
verbose: TJBot response from conversation workspace id 7e9bd2c5-5136-40bb-9388-6c5299440638 Let me have a look :
verbose: TJBot speaking with voice en-US_MichaelV2Voice
verbose: TJBot taking a photo
debug: capturing image at path: /tmp/tjbot202087-1512-1vwe86j.mudsi
debug: sending image to Watson Visual Recognition
debug: wrote audio stream to temp file /tmp/tjbot202087-1512-q2dig0.brt6
verbose: TJBot speaking: Let me have a look
debug: listening paused
debug: microphone paused
debug: Playing audio with parameters:  filename=/tmp/tjbot202087-1512-q2dig0.brt6, gain=100, debug=true, player=aplay, device=plughw:0,0
========= { filename: '/tmp/tjbot202087-1512-q2dig0.brt6',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:0,0' }
verbose: TJBot speaking with voice en-US_MichaelV2Voice
debug: audio playback finished
debug: listening resumed
debug: wrote audio stream to temp file /tmp/tjbot202087-1512-1mctdk.ozybhk
verbose: TJBot speaking: I'm looking at belting, fabric, and space bar, and a few other things too
debug: listening paused
debug: microphone paused
debug: Playing audio with parameters:  filename=/tmp/tjbot202087-1512-1mctdk.ozybhk, gain=100, debug=true, player=aplay, device=plughw:0,0
========= { filename: '/tmp/tjbot202087-1512-1mctdk.ozybhk',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:0,0' }
debug: audio playback finished
debug: listening resumed
```

erramilli, 2020-09-07:
@chunyianliew  Thank you so much..
I'm working on your suggestions. 

One thing I noticed when I tried to do a npm install .. I noticed this which I missed last time.
pi@Pbot:~/Desktop/tjbot/recipes/conversation $ npm install
**npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we**
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/
up to date in 10.534s

is it posssible this is the root cause of my problem ?

chunyianliew, 2020-09-07:
It wouldn't hurt to use a supported Node version when troubleshooting issues like you have run into:
Here you can find instructions how to downgrade the Node version:
https://github.com/ibmtjbot/tjbot/issues/153#issuecomment-580382547

erramilli, 2020-09-07:
@chunyianliew : is it possible to send your  config.js (possibly changing the IDs a little bit), I know it may be asking for too much..

erramilli, 2020-09-07:
@chunyianliew :  Thank you so so much for your help and prompt responses.. it turns out to be dumb mistake on my side as I was looking at the Config.js  file and program was looking for  just 'config' i.e with no extension. 

The conversation works fine now. I'm so excited now  :+1:

closed 2020-09-07 by erramilli

chunyianliew, 2020-09-07:
@erramilli oh, really? That is a bit strange, because the config file is supposed to be named config.js.
But I am glad that you figured it out what caused the issue.