ibmtjbot/tjbot #162: Visual Recognition Error 400 (vmodel-id header must be non-empty)

erramilli, 2020-09-09:
I just got a camera for my Tjbot and running the conversation.js 

When I try to use a command that involves Camera..  e.g. 'Tjbot What do you see',  I can see the camera starting and doing something..then throws the following error.

any idea on what is wrong..and how to fix it?

Unhandled rejection Error: {"images":[{"image":"tjbot202088-1792-aakn6t.0f78m.jpg","error":{"code":400,"description":"Invalid Request: provided vmodel-id header must be non-empty"}}],"images_processed":1}
    at Request._callback (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/requestwrapper.js:113:21)
    at Request.self.callback (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:185:22)
    at emitTwo (events.js:126:13)
    at Request.emit (events.js:214:7)
    at Request.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:1157:10)
    at emitOne (events.js:116:13)
    at Request.emit (events.js:211:7)
    at IncomingMessage.<anonymous> (/home/pi/Desktop/tjbot/recipes/conversation/node_modules/request/request.js:1079:12)
    at Object.onceWrapper (events.js:313:30)
    at emitNone (events.js:111:20)
    at IncomingMessage.emit (events.js:208:7)
    at endReadableNT (_stream_readable.js:1064:12)
    at _combinedTickCallback (internal/process/next_tick.js:139:11)
    at process._tickCallback (internal/process/next_tick.js:181:9)

chunyianliew, 2020-09-09:
You can fix this by adjusting one file, lib/tjbot.js
For the conversation recipe that file is located here: :~/Desktop/tjbot/recipes/conversation/node_modules/tjbot/lib/tjbot.js
You need to change one line as shown in the following pull request:
https://github.com/ibmtjbot/tjbotlib/pull/59/commits/e57e52b7fdf21bef97eee9a45773b359804285ae

erramilli, 2020-09-11:
thank you so much.. now to the next problem..
When I say 'What do you see' , the camera turns on...and then gives the below error. And the red light on the camera stays on.

error: Error taking picture. message=2020-09-11T22:17:14.128Z Raspistill failed: No file found, stack=Error: 2020-09-11T22:17:14.128Z Raspistill failed: No file found
    at /home/pi/Desktop/tjbot/recipes/conversation/node_modules/node-raspistill/build/lib/camera/default.js:110:19
    at <anonymous>
Unhandled rejection Error: 2020-09-11T22:17:14.128Z Raspistill failed: No file found
    at /home/pi/Desktop/tjbot/recipes/conversation/node_modules/node-raspistill/build/lib/camera/default.js:110:19
    at <anonymous>

chunyianliew, 2020-09-14:
Can you please verify if the program required to take a picture has been installed on your tjbot?

Here you can find more detailed instructions how to install a pi camera and how to take a picture from the shell:
https://raspberrytips.com/install-camera-raspberry-pi/

erramilli, 2020-09-14:
@chunyianliew Thank you so much for all your help.. 

it turns out that the camera is faulty.. Amazon gave the option for a free replacement and I got it just now.
It works just fine ...
Thank you so much for your help with the original error.

closed 2020-09-14 by erramilli