ibmtjbot/tjbot #163: Issue with Speech-to-text, possibly USB microphone

DaveAtwood, 2020-09-14:
When I attempt to run the stt.js recipe I get this:

`pi@Datwood:~/Desktop/tjbot/recipes/speech_to_text $ sudo node stt.js 
info: Hello from TJBot! My name is TJ.
I understand lots of colors.  You can tell me to shine my light a different color by saying 'turn the light red' or 'change the light to green' or 'turn the light off'.
(node:1174) Warning: N-API is an experimental feature and could change at any time.
pi@Datwood:~/Desktop/tjbot/recipes/speech_to_text $`

It immediately drops me out of the run time, not giving me any time to ask TJBot to change color.

I added:
```var tjConfig = {
    verboseLogging: true,
    microphoneDeviceId: 'plughw:2,0',```

to the stt.js file based on information I got from the `aplay -l` and `arecord -l` commands (listed below). I tried `plughw:1,0` also just to see if it would work (it didn't)

Running:
* node v9.11.2
* npm 5.6.0

**aplay -l**
`pi@Datwood:~/Desktop/tjbot/recipes/speech_to_text $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3`

**arecord -l**
`pi@Datwood:~/Desktop/tjbot/recipes/speech_to_text $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

If anyone has any ideas please let me know! Thanks

DaveAtwood, 2020-09-14:
I also tried using this block. It gave a more verbose output, but still kicked me out of the runtime without listening to the words I was saying

var mic = require('mic');
var fs = require('fs');

var micInstance = mic({ 'rate': '44100', 'channels': '2', 'debug': false, 'exitOnSilence': 6, 'device': 'plughw:1,0' });

var micInputStream = micInstance.getAudioStream();
 
var outputFileStream = fs.WriteStream('output.raw');
 
micInputStream.pipe(outputFileStream);
 
micInputStream.on('data', function(data) {
    console.log("Recieved Input Stream: " + data.length);

var micInputStream = micInstance.getAudioStream();
 
var outputFileStream = fs.WriteStream('output.raw');
 
micInputStream.pipe(outputFileStream);
 
micInputStream.on('data', function(data) {
    console.log("Recieved Input Stream: " + data.length);
});
 
micInputStream.on('error', function(err) {
    cosole.log("Error in Input Stream: " + err);
});
 
micInputStream.on('startComplete', function() {
    console.log("Got SIGNAL startComplete");
    setTimeout(function() {
            micInstance.pause();

micInputStream.on('error', function(err) {
    cosole.log("Error in Input Stream: " + err);
});
 
micInputStream.on('startComplete', function() {
    console.log("Got SIGNAL startComplete");
    setTimeout(function() {
            micInstance.pause();
    }, 5000);
});
    
micInputStream.on('stopComplete', function() {
    console.log("Got SIGNAL stopComplete");
});
    
micInputStream.on('pauseComplete', function() {
    console.log("Got SIGNAL pauseComplete");
    setTimeout(function() {
        micInstance.resume();
    }, 5000);
});
 
micInputStream.on('resumeComplete', function() {
    console.log("Got SIGNAL resumeComplete");
    setTimeout(function() {
        micInstance.stop();
    }, 5000);
});

micInputStream.on('silence', function() {
    console.log("Got SIGNAL silence");
});
 
micInputStream.on('processExitComplete', function() {
    console.log("Got SIGNAL processExitComplete");
});
 
micInstance.start();

chunyianliew, 2020-09-14:
Based on the output of "arecord -l" it shows that your USB mic is being identified as card 2 and device 0.
Please try to record 4 seconds of you speaking in to the mic:

`pi@tjbot:~/Desktop/tjbot/recipes/speech_to_text $ arecord -f S16_LE -r 16000 -D plughw:2,0 -d 4 test.wav
`
And play the test file to verify that the mic is has worked correctly, using 1,0 for plughw assuming that you are using the 3.5mm audio jack for sound as no usb speaker is listed for "aplay -l":

`pi@tjbot:~/Desktop/tjbot/recipes/speech_to_text $ aplay -vv test.wav -D plughw:1,0
`
If this all works fine, I can only assume that you followed the instructions as documented here: https://github.com/ibmtjbot/tjbot/tree/master/recipes/speech_to_text
Can you please share the output of the "npm install" command?

CaskAle, 2020-09-18:
this issue is related the board revision.  I have experienced other issues around this.  The pi 4 has had an engineering change and there is an additional revision code.  The stt.js app works on one revision but not the other.  I am still looking as to why.

AilinLuca, 2020-11-14:
Hello! I had the same issue. I ran arecord and no sound was picked up; the file contained 4 seconds of silence. npm install resulted in "up to date, audited 331 packages in 5s".

Digging into it, I think the problem is I'm running it headless on my mac and VNC doesn't support audio.