ibmtjbot/tjbot #164: Broken test suite

TheVinhLuong102, 2020-10-13:
Hi TJBot repo owners,

I have installed the TJBot software on a Raspberry Pi 3  by running `curl -sL http://ibm.biz/tjbot-bootstrap | sudo sh -`. When I run the tests in `~/Desktop/tjbot/bootstrap`, though, I encounter the following errors:

```
sudo ./runTests.sh 
Installing support libraries for TJBot. This may take few minutes.
Running camera test
internal/modules/cjs/loader.js:638
  throw err;
  ^
Error: Cannot find module ‘tjbot’
  at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
  at Function.Module._load (internal/modules/cjs/loader.js:562:25)
  at Module.require (internal/modules/cjs/loader.js:692:17)
  at require (internal/modules/cjs/helpers.js:25:18)
  at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/test.camera.js:21:15)
  at Module._compile (internal/modules/cjs/loader.js:778:30)
  at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
  at Module.load (internal/modules/cjs/loader.js:653:32)
  at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
  at Function.Module._load (internal/modules/cjs/loader.js:585:3)
Running LED test
internal/modules/cjs/loader.js:638
  throw err;
  ^
Error: Cannot find module ‘tjbot’
  at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
  at Function.Module._load (internal/modules/cjs/loader.js:562:25)
  at Module.require (internal/modules/cjs/loader.js:692:17)
  at require (internal/modules/cjs/helpers.js:25:18)
  at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/test.led.js:21:15)
  at Module._compile (internal/modules/cjs/loader.js:778:30)
  at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
  at Module.load (internal/modules/cjs/loader.js:653:32)
  at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
  at Function.Module._load (internal/modules/cjs/loader.js:585:3)
Running servo test
internal/modules/cjs/loader.js:638
  throw err;
  ^
Error: Cannot find module ‘tjbot’
  at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
  at Function.Module._load (internal/modules/cjs/loader.js:562:25)
  at Module.require (internal/modules/cjs/loader.js:692:17)
  at require (internal/modules/cjs/helpers.js:25:18)
  at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/test.servo.js:21:15)
  at Module._compile (internal/modules/cjs/loader.js:778:30)
  at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
  at Module.load (internal/modules/cjs/loader.js:653:32)
  at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
  at Function.Module._load (internal/modules/cjs/loader.js:585:3)
Running speaker test
internal/modules/cjs/loader.js:638
  throw err;
  ^
Error: Cannot find module ‘tjbot’
  at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
  at Function.Module._load (internal/modules/cjs/loader.js:562:25)
  at Module.require (internal/modules/cjs/loader.js:692:17)
  at require (internal/modules/cjs/helpers.js:25:18)
  at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/test.speaker.js:21:15)
  at Module._compile (internal/modules/cjs/loader.js:778:30)
  at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
  at Module.load (internal/modules/cjs/loader.js:653:32)
  at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
  at Function.Module._load (internal/modules/cjs/loader.js:585:3)
```

Does this mean the test suite is broken? Or did I miss out something critical?

Thank you very much!

chunyianliew, 2020-10-13:
Did you install the dependencies as explained here under "Running hardware tests"?: https://github.com/ibmtjbot/tjbot/blob/master/bootstrap/README.md

TheVinhLuong102, 2020-10-13:
@chunyianliew yes I ran the bootstrap script, as mentioned above. Does that bootstrapping still miss out something important?

chunyianliew, 2020-10-13:
You also need to install the dependencies as explained almost at the bottom of the page under the title "Running hardware tests":

`npm install ~/Desktop/tjbot/bootstrap/tests
`

TheVinhLuong102, 2020-10-13:
@chunyianliew ok thank you for letting me know. It's not very clearly documented in the top-level documentation, so I didn't know those had to be installed first. The tests do run now, albeit not all successfully. Let me close this issue now.

TheVinhLuong102, 2020-10-15:
@chunyianliew the tests keep failing, so I'm reopening this ticket.

My environment info is as follows:
- Raspberry Pi 3 Model B+
- Raspbian Stretch (version 2019-04-08)
- npm 5.6.0
- nodejs v9.11.2

The output of my various commands are as follows:

```
npm install ~/Desktop/tjbot/bootstrap/tests

npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ tests@1.0.0
updated 1 package in 13.838s
```

```
sudo ./runTests.sh 

Installing support libraries for TJBot. This may take few minutes.
Running camera test
internal/modules/cjs/loader.js:550
    throw err;
    ^

Error: Cannot find module './build/Release/node_sleep.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:548:15)
    at Function.Module._load (internal/modules/cjs/loader.js:475:25)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/sleep/index.js:1:75)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/tjbot/lib/tjbot.js:24:15)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
Running LED test
internal/modules/cjs/loader.js:550
    throw err;
    ^

Error: Cannot find module './build/Release/node_sleep.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:548:15)
    at Function.Module._load (internal/modules/cjs/loader.js:475:25)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/sleep/index.js:1:75)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/tjbot/lib/tjbot.js:24:15)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
Running servo test
internal/modules/cjs/loader.js:550
    throw err;
    ^

Error: Cannot find module './build/Release/node_sleep.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:548:15)
    at Function.Module._load (internal/modules/cjs/loader.js:475:25)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/sleep/index.js:1:75)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/tjbot/lib/tjbot.js:24:15)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
Running speaker test
internal/modules/cjs/loader.js:550
    throw err;
    ^

Error: Cannot find module './build/Release/node_sleep.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:548:15)
    at Function.Module._load (internal/modules/cjs/loader.js:475:25)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/sleep/index.js:1:75)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/Desktop/tjbot/bootstrap/tests/node_modules/tjbot/lib/tjbot.js:24:15)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
-------------------------------------------------------------------
Tests complete. Have fun! ;)
-------------------------------------------------------------------
```

Could you let me know what could be behind these errors?

Thank you very much @chunyianliew.

jweisz, 2020-11-13:
Closing this issue as the TJBot 2.0.0 release should resolve all outstanding issues with the hardware tests (which requires Node v15.x).

closed 2020-11-13 by jweisz