ibmtjbot/tjbot #170: TJBot 2.0.0: Broken Tests

TheVinhLuong102, 2020-11-15:
@jweisz I am trying out the newly-updated TJBot library v2.0.0 but of most of the tests are still failing. 

My NodeJS and NPM versions are as follows:
> ~/Desktop/tjbot/tests $ nodejs -v
> v15.2.0
> ~/Desktop/tjbot/tests $ npm -v
> 7.0.8

I have installed the test dependencies:
> npm install ~/Desktop/tjbot/tests
> up to date, audited 331 packages in 6s

In the below comments, I'm pasting the errors/warnings from the 5 tests:

TheVinhLuong102, 2020-11-15:
> sudo nodejs test.camera.js

[rpi-ws281x-native] Could not verify raspberry-pi version. If this is wrong and you are running this on a raspberry-pi, please file a bug-report at   https://github.com/beyondscreen/node-rpi-ws281x-native/issues
A non-functional stub of this modules interface will be returned.
+-----------------------------------------------------------------------+
| Warning: The pigpio C library can't be loaded on this machine and any |
| attempt to use it will fail.                                          |
|                                                                       |
| Error: "Module did not self-register"                                 |
| ------------------------------------                                  |
| If you are working on a Raspberry Pi and see a "Module did not self-  |
| register" error, this typically indicates that the installation       |
| instructions were not exactly followed. For further details see the   |
| installation section of the readme at                                 |
| https://github.com/fivdi/pigpio#installation                          |
| Note that step 1 of the installation instructions must be completed   |
| before step 2.                                                        |
+-----------------------------------------------------------------------+

Invoking require('bindings')('pigpio.node') resulted in the follwoing error:

Error: Module did not self-register: '/home/pi/Desktop/tjbot/tests/node_modules/pigpio/build/Release/pigpio.node'.
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1161:18)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at Module.require (node:internal/modules/cjs/loader:991:19)
    at require (node:internal/modules/cjs/helpers:92:18)
    at bindings (/home/pi/Desktop/tjbot/tests/node_modules/bindings/bindings.js:112:48)
    at /home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:8:31
    at Object.<anonymous> (/home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:43:3)
    at Module._compile (node:internal/modules/cjs/loader:1102:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1131:10)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:197:29)
    at ModuleJob.run (node:internal/modules/esm/module_job:146:23)
    at async Loader.import (node:internal/modules/esm/loader:165:24)
    at async Object.loadESM (node:internal/process/esm_loader:68:5)

-------------------------------------------------------------------------------

info: Hello from TJBot!
verbose: TJBot library version v2.0.0
silly: TJBot configuration: {"log":{"level":"silly"},"robot":{"gender":"male"},"converse":{},"listen":{"microphoneDeviceId":"plughw:1,0","inactivityTimeout":-1,"backgroundAudioSuppression":0.4,"language":"en-US"},"wave":{"servoPin":7},"speak":{"language":"en-US","speakerDeviceId":"plughw:0,0"},"see":{"confidenceThreshold":0.6,"camera":{"height":720,"width":960,"verticalFlip":false,"horizontalFlip":false}},"shine":{"neopixel":{"gpioPin":18,"grbFormat":false},"commonAnode":{"redPin":19,"greenPin":13,"bluePin":12}}}
info: Initializing TJBot with camera
verbose: initializing camera
verbose: capturing image at path: /tmp/tjbot20201015-6118-1dm3zep.h7j7.jpg
silly: image path: /tmp, image filename: tjbot20201015-6118-1dm3zep.h7j7
silly: camera options: {"noFileSave":false,"verticalFlip":false,"horizontalFlip":false,"noPreview":true,"outputDir":"/tmp","encoding":"jpg","width":960,"height":720,"time":1,"fileName":"tjbot20201015-6118-1dm3zep.h7j7"}
Please note this test will only work from a local terminal window on your
Raspberry Pi and will not display an image when run via SSH.
Did TJBot show a picture from the camera (Y/n)? n
file:///home/pi/Desktop/tjbot/tests/test.camera.js:29
        throw new Error(`TJBot did not ${behavior}`);
              ^

Error: TJBot did not show a picture from the camera
    at confirm (file:///home/pi/Desktop/tjbot/tests/test.camera.js:29:15)
    at file:///home/pi/Desktop/tjbot/tests/test.camera.js:50:1

TheVinhLuong102, 2020-11-15:
> sudo nodejs test.led.js 

[rpi-ws281x-native] Could not verify raspberry-pi version. If this is wrong and you are running this on a raspberry-pi, please file a bug-report at   https://github.com/beyondscreen/node-rpi-ws281x-native/issues
A non-functional stub of this modules interface will be returned.
+-----------------------------------------------------------------------+
| Warning: The pigpio C library can't be loaded on this machine and any |
| attempt to use it will fail.                                          |
|                                                                       |
| Error: "Module did not self-register"                                 |
| ------------------------------------                                  |
| If you are working on a Raspberry Pi and see a "Module did not self-  |
| register" error, this typically indicates that the installation       |
| instructions were not exactly followed. For further details see the   |
| installation section of the readme at                                 |
| https://github.com/fivdi/pigpio#installation                          |
| Note that step 1 of the installation instructions must be completed   |
| before step 2.                                                        |
+-----------------------------------------------------------------------+

Invoking require('bindings')('pigpio.node') resulted in the follwoing error:

Error: Module did not self-register: '/home/pi/Desktop/tjbot/tests/node_modules/pigpio/build/Release/pigpio.node'.
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1161:18)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at Module.require (node:internal/modules/cjs/loader:991:19)
    at require (node:internal/modules/cjs/helpers:92:18)
    at bindings (/home/pi/Desktop/tjbot/tests/node_modules/bindings/bindings.js:112:48)
    at /home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:8:31
    at Object.<anonymous> (/home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:43:3)
    at Module._compile (node:internal/modules/cjs/loader:1102:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1131:10)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:197:29)
    at ModuleJob.run (node:internal/modules/esm/module_job:146:23)
    at async Loader.import (node:internal/modules/esm/loader:165:24)
    at async Object.loadESM (node:internal/process/esm_loader:68:5)

-------------------------------------------------------------------------------

info: Hello from TJBot!
verbose: TJBot library version v2.0.0
silly: TJBot configuration: {"log":{"level":"silly"},"robot":{"gender":"male"},"converse":{},"listen":{"microphoneDeviceId":"plughw:1,0","inactivityTimeout":-1,"backgroundAudioSuppression":0.4,"language":"en-US"},"wave":{"servoPin":7},"speak":{"language":"en-US","speakerDeviceId":"plughw:0,0"},"see":{"confidenceThreshold":0.6,"camera":{"height":720,"width":960,"verticalFlip":false,"horizontalFlip":false}},"shine":{"neopixel":{"gpioPin":18,"grbFormat":false},"commonAnode":{"redPin":19,"greenPin":13,"bluePin":12}}}
Are you testing a Neopixel LED (n) or Common Anode (c) LED, or both (b)? (N/c/b)? 
testing Neopixel LED
info: Initializing TJBot with led_neopixel
verbose: initializing led_neopixel on PIN 18
verbose: shining my LED to RGB color 0xFF0000
Did TJBot shine the light red (Y/n)? n
file:///home/pi/Desktop/tjbot/tests/test.led.js:27
        throw new Error(`TJBot did not ${behavior}`);
              ^

Error: TJBot did not shine the light red
    at confirm (file:///home/pi/Desktop/tjbot/tests/test.led.js:27:15)
    at file:///home/pi/Desktop/tjbot/tests/test.led.js:49:5
    at Array.forEach (<anonymous>)
    at file:///home/pi/Desktop/tjbot/tests/test.led.js:47:8
    at ModuleJob.run (node:internal/modules/esm/module_job:146:23)
    at async Loader.import (node:internal/modules/esm/loader:165:24)
    at async Object.loadESM (node:internal/process/esm_loader:68:5)

TheVinhLuong102, 2020-11-15:
For the Microphone test, where should we put the `ibm-credentials.env` file?

> sudo nodejs test.mic.js

[rpi-ws281x-native] Could not verify raspberry-pi version. If this is wrong and you are running this on a raspberry-pi, please file a bug-report at   https://github.com/beyondscreen/node-rpi-ws281x-native/issues
A non-functional stub of this modules interface will be returned.
+-----------------------------------------------------------------------+
| Warning: The pigpio C library can't be loaded on this machine and any |
| attempt to use it will fail.                                          |
|                                                                       |
| Error: "Module did not self-register"                                 |
| ------------------------------------                                  |
| If you are working on a Raspberry Pi and see a "Module did not self-  |
| register" error, this typically indicates that the installation       |
| instructions were not exactly followed. For further details see the   |
| installation section of the readme at                                 |
| https://github.com/fivdi/pigpio#installation                          |
| Note that step 1 of the installation instructions must be completed   |
| before step 2.                                                        |
+-----------------------------------------------------------------------+

Invoking require('bindings')('pigpio.node') resulted in the follwoing error:

Error: Module did not self-register: '/home/pi/Desktop/tjbot/tests/node_modules/pigpio/build/Release/pigpio.node'.
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1161:18)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at Module.require (node:internal/modules/cjs/loader:991:19)
    at require (node:internal/modules/cjs/helpers:92:18)
    at bindings (/home/pi/Desktop/tjbot/tests/node_modules/bindings/bindings.js:112:48)
    at /home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:8:31
    at Object.<anonymous> (/home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:43:3)
    at Module._compile (node:internal/modules/cjs/loader:1102:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1131:10)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:197:29)
    at ModuleJob.run (node:internal/modules/esm/module_job:146:23)
    at async Loader.import (node:internal/modules/esm/loader:165:24)
    at async Object.loadESM (node:internal/process/esm_loader:68:5)

-------------------------------------------------------------------------------

info: Hello from TJBot!
verbose: TJBot library version v2.0.0
silly: TJBot configuration: {"log":{"level":"silly"},"robot":{"gender":"male"},"converse":{},"listen":{"microphoneDeviceId":"plughw:1,0","inactivityTimeout":-1,"backgroundAudioSuppression":0.4,"language":"en-US"},"wave":{"servoPin":7},"speak":{"language":"en-US","speakerDeviceId":"plughw:0,0"},"see":{"confidenceThreshold":0.6,"camera":{"height":720,"width":960,"verticalFlip":false,"horizontalFlip":false}},"shine":{"neopixel":{"gpioPin":18,"grbFormat":false},"commonAnode":{"redPin":19,"greenPin":13,"bluePin":12}}}
info: Initializing TJBot with microphone
verbose: initializing microphone
Note: This test requires credentials for the Speech to Text service stored in the ibm-credentials.env file.
If you see an error about "Missing required parameters: apikey", it is because TJBot was unable
to find credentials for Speech to Text.

TJBot is listening and will echo what you say on the console. Say "stop" to stop the test.
verbose: initializing speech_to_text service
/home/pi/Desktop/tjbot/tests/node_modules/ibm-cloud-sdk-core/lib/helper.js:102
        ? new Error('Missing required parameters: ' + missing.join(', '))
          ^

Error: Missing required parameters: apikey
    at Object.getMissingParams (/home/pi/Desktop/tjbot/tests/node_modules/ibm-cloud-sdk-core/lib/helper.js:102:11)
    at Object.validateInput (/home/pi/Desktop/tjbot/tests/node_modules/ibm-cloud-sdk-core/auth/utils/helpers.js:66:39)
    at new IamAuthenticator (/home/pi/Desktop/tjbot/tests/node_modules/ibm-cloud-sdk-core/auth/authenticators/iam-authenticator.js:68:17)
    at Object.getAuthenticatorFromEnvironment (/home/pi/Desktop/tjbot/tests/node_modules/ibm-cloud-sdk-core/auth/utils/get-authenticator-from-environment.js:75:29)
    at SpeechToTextV1 [as constructor] (/home/pi/Desktop/tjbot/tests/node_modules/ibm-watson/speech-to-text/v1-generated.js:73:58)
    at new SpeechToTextV1 (/home/pi/Desktop/tjbot/tests/node_modules/ibm-watson/speech-to-text/v1.js:42:23)
    at TJBot._createServiceAPI (file:///home/pi/Desktop/tjbot/tests/node_modules/tjbot/lib/tjbot.js:484:25)
    at TJBot._assertCapability (file:///home/pi/Desktop/tjbot/tests/node_modules/tjbot/lib/tjbot.js:556:22)
    at TJBot.listen (file:///home/pi/Desktop/tjbot/tests/node_modules/tjbot/lib/tjbot.js:793:14)
    at file:///home/pi/Desktop/tjbot/tests/test.mic.js:30:29
    at ModuleJob.run (node:internal/modules/esm/module_job:146:23)
    at async Loader.import (node:internal/modules/esm/loader:165:24)
    at async Object.loadESM (node:internal/process/esm_loader:68:5)

TheVinhLuong102, 2020-11-15:
> sudo nodejs test.servo.js

[rpi-ws281x-native] Could not verify raspberry-pi version. If this is wrong and you are running this on a raspberry-pi, please file a bug-report at   https://github.com/beyondscreen/node-rpi-ws281x-native/issues
A non-functional stub of this modules interface will be returned.
+-----------------------------------------------------------------------+
| Warning: The pigpio C library can't be loaded on this machine and any |
| attempt to use it will fail.                                          |
|                                                                       |
| Error: "Module did not self-register"                                 |
| ------------------------------------                                  |
| If you are working on a Raspberry Pi and see a "Module did not self-  |
| register" error, this typically indicates that the installation       |
| instructions were not exactly followed. For further details see the   |
| installation section of the readme at                                 |
| https://github.com/fivdi/pigpio#installation                          |
| Note that step 1 of the installation instructions must be completed   |
| before step 2.                                                        |
+-----------------------------------------------------------------------+

Invoking require('bindings')('pigpio.node') resulted in the follwoing error:

Error: Module did not self-register: '/home/pi/Desktop/tjbot/tests/node_modules/pigpio/build/Release/pigpio.node'.
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1161:18)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at Module.require (node:internal/modules/cjs/loader:991:19)
    at require (node:internal/modules/cjs/helpers:92:18)
    at bindings (/home/pi/Desktop/tjbot/tests/node_modules/bindings/bindings.js:112:48)
    at /home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:8:31
    at Object.<anonymous> (/home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:43:3)
    at Module._compile (node:internal/modules/cjs/loader:1102:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1131:10)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:197:29)
    at ModuleJob.run (node:internal/modules/esm/module_job:146:23)
    at async Loader.import (node:internal/modules/esm/loader:165:24)
    at async Object.loadESM (node:internal/process/esm_loader:68:5)

-------------------------------------------------------------------------------

info: Hello from TJBot!
verbose: TJBot library version v2.0.0
silly: TJBot configuration: {"log":{"level":"silly"},"robot":{"gender":"male"},"converse":{},"listen":{"microphoneDeviceId":"plughw:1,0","inactivityTimeout":-1,"backgroundAudioSuppression":0.4,"language":"en-US"},"wave":{"servoPin":7},"speak":{"language":"en-US","speakerDeviceId":"plughw:0,0"},"see":{"confidenceThreshold":0.6,"camera":{"height":720,"width":960,"verticalFlip":false,"horizontalFlip":false}},"shine":{"neopixel":{"gpioPin":18,"grbFormat":false},"commonAnode":{"redPin":19,"greenPin":13,"bluePin":12}}}
info: Initializing TJBot with servo
verbose: initializing servo on PIN 7
Moving TJBot's arm back
file:///home/pi/Desktop/tjbot/tests/node_modules/tjbot/lib/tjbot.js:606
                throw new Error(
                      ^

Error: TJBot is not configured with an arm. Please check that you included the servo hardware in the TJBot initialize() method.
    at TJBot._assertCapability (file:///home/pi/Desktop/tjbot/tests/node_modules/tjbot/lib/tjbot.js:606:23)
    at TJBot.armBack (file:///home/pi/Desktop/tjbot/tests/node_modules/tjbot/lib/tjbot.js:1667:14)
    at file:///home/pi/Desktop/tjbot/tests/test.servo.js:35:7
    at ModuleJob.run (node:internal/modules/esm/module_job:146:23)
    at async Loader.import (node:internal/modules/esm/loader:165:24)
    at async Object.loadESM (node:internal/process/esm_loader:68:5)

TheVinhLuong102, 2020-11-15:
The Speaker test is the only one that succeeds, but still outputs lots of warnings:

> sudo nodejs test.speaker.js

[rpi-ws281x-native] Could not verify raspberry-pi version. If this is wrong and you are running this on a raspberry-pi, please file a bug-report at   https://github.com/beyondscreen/node-rpi-ws281x-native/issues
A non-functional stub of this modules interface will be returned.
+-----------------------------------------------------------------------+
| Warning: The pigpio C library can't be loaded on this machine and any |
| attempt to use it will fail.                                          |
|                                                                       |
| Error: "Module did not self-register"                                 |
| ------------------------------------                                  |
| If you are working on a Raspberry Pi and see a "Module did not self-  |
| register" error, this typically indicates that the installation       |
| instructions were not exactly followed. For further details see the   |
| installation section of the readme at                                 |
| https://github.com/fivdi/pigpio#installation                          |
| Note that step 1 of the installation instructions must be completed   |
| before step 2.                                                        |
+-----------------------------------------------------------------------+

Invoking require('bindings')('pigpio.node') resulted in the follwoing error:

Error: Module did not self-register: '/home/pi/Desktop/tjbot/tests/node_modules/pigpio/build/Release/pigpio.node'.
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1161:18)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at Module.require (node:internal/modules/cjs/loader:991:19)
    at require (node:internal/modules/cjs/helpers:92:18)
    at bindings (/home/pi/Desktop/tjbot/tests/node_modules/bindings/bindings.js:112:48)
    at /home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:8:31
    at Object.<anonymous> (/home/pi/Desktop/tjbot/tests/node_modules/pigpio/pigpio.js:43:3)
    at Module._compile (node:internal/modules/cjs/loader:1102:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1131:10)
    at Module.load (node:internal/modules/cjs/loader:967:32)
    at Function.Module._load (node:internal/modules/cjs/loader:807:14)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:197:29)
    at ModuleJob.run (node:internal/modules/esm/module_job:146:23)
    at async Loader.import (node:internal/modules/esm/loader:165:24)
    at async Object.loadESM (node:internal/process/esm_loader:68:5)

-------------------------------------------------------------------------------

info: Hello from TJBot!
verbose: TJBot library version v2.0.0
silly: TJBot configuration: {"log":{"level":"silly"},"robot":{"gender":"male"},"converse":{},"listen":{"microphoneDeviceId":"plughw:1,0","inactivityTimeout":-1,"backgroundAudioSuppression":0.4,"language":"en-US"},"wave":{"servoPin":7},"speak":{"language":"en-US","speakerDeviceId":"plughw:0,0"},"see":{"confidenceThreshold":0.6,"camera":{"height":720,"width":960,"verticalFlip":false,"horizontalFlip":false}},"shine":{"neopixel":{"gpioPin":18,"grbFormat":false},"commonAnode":{"redPin":19,"greenPin":13,"bluePin":12}}}
info: Initializing TJBot with speaker
verbose: initializing speaker
silly: playing audio with parameters:  {"filename":"/usr/share/sounds/alsa/Front_Center.wav","gain":100,"debug":true,"player":"aplay","device":"plughw:0,0"}
========= {
  filename: '/usr/share/sounds/alsa/Front_Center.wav',
  gain: 100,
  debug: true,
  player: 'aplay',
  device: 'plughw:0,0'
}
silly: audio playback finished
Did TJBot play the "Front Center" sound (Y/n)? Y

jweisz, 2020-11-16:
Hi @TheVinhLuong102 -- it looks like you have a few things going on:

- There seems to be an issue with `pigpio`, which is why you're seeing lots of error messages. I'd say first try wiping out & reinstalling the node modules: `rm -rf node_modules && npm install`. Sometimes this helps. If not, you might need to do `sudo apt-get install pigpio`, though I was under the impression that this package is already installed on recent Raspian releases.
- Another way to debug the `pigpio` error is to try making a new node app and directly installing `pigpio` to see what errors it generates.
```
cd ~/Desktop
mkdir gpiotest
cd gpiotest
npm init
npm install --save pigpio
```
- For the mic test, put the `ibm-credentials.env` file in the `tests/` directory (e.g. at the same level as `test.mic.js`).
- For the speaker test, the warnings are just complaining about `pigpio`

TheVinhLuong102, 2020-11-16:
@jweisz thank you for the suggestions.

Following your comments, I did `sudo apt-get install pigpio` and then `rm -rf node_modules && npm install` in the tests dir and the `pigpio` errors have gone away.

The other tests still have problems:

- LED test: the LEDs don't turn on at all. I'm using the LEDs provided by AdaFruit/SparkFun, and with a RPi 4.

- Mic Test: I've put in an `ibm-credentials.env` file in the `tests` folder but the Mic test still fails to find the credentials:
> If you see an error about "Missing required parameters: apikey", it is because TJBot was unable to find credentials for Speech to Text.
> TJBot is listening and will echo what you say on the console. [...]

My `ibm-credentials.env` file has the following format -- is it the right format?
> ASSISTANT_API_KEY=...
> ASSISTANT_API_ENDPOINT=...
> ASSISTANT_WORKSPACE_ID=
> LANGUAGE_TRANSLATOR_API_KEY=...
> LANGUAGE_TRANSLATOR_API_ENDPOINT=...
> SPEECH_TO_TEXT_API_KEY=...
> SPEECH_TO_TEXT_API_ENDPOINT=...
> TEXT_TO_SPEECH_API_KEY=...
> TEXT_TO_SPEECH_API_ENDPOINT=...
> TONE_ANALYZER_API_KEY=...
> TONE_ANALYZER_API_ENDPOINT=...
> VISUAL_RECOGNITION_API_KEY=...
> VISUAL_RECOGNITION_API_ENDPOINT=...

- Servo test: the servo did move but the motions are not those indicated in the test, e.g. Backward, Raise, Lower.