ibmtjbot/tjbot #172: TJbot, assistantId error.

MUbarak123-56, 2020-11-17:
I have plugged in my assistantId into the config.js file but when I ran sudo node conversation.js and successfully passed a command like "Watson Hello," I got the error below:

error: error creating session for assistant service. please check that tj.configuration.converse.assistantId is defined.
/home/pi/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/request-wrapper.js:224
        var error = new Error();
                    ^

Not Found: Resource not found
    at RequestWrapper.formatError (/home/pi/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/request-wrapper.js:224:21)
    at /home/pi/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/request-wrapper.js:212:25
    at processTicksAndRejections (node:internal/process/task_queues:93:5) {
  statusText: 'Not Found',
  status: 404,
  code: 404,
  body: '{"error":"Resource not found","code":404}',
  headers: {
    'content-type': 'application/json; charset=utf-8',
    'content-length': '41',
    'access-control-allow-origin': '*',
    'access-control-allow-methods': 'GET, POST, PUT, DELETE, PATCH, HEAD, OPTIONS',
    'access-control-allow-headers': 'Content-Type, Content-Length, Authorization, X-Watson-Authorization-Token, X-WDC-PL-OPT-OUT, X-Watson-UserInfo, X-Watson-Learning-Opt-Out, X-Watson-Metadata',
    'access-control-max-age': '3600',
    'content-security-policy': "default-src 'none'",
    'x-dns-prefetch-control': 'off',
    'x-frame-options': 'SAMEORIGIN',
    'strict-transport-security': 'max-age=31536000; includeSubDomains;',
    'x-download-options': 'noopen',
    'x-content-type-options': 'nosniff',
    'x-xss-protection': '1; mode=block',
    'x-response-time': '8.473ms',
    'x-global-transaction-id': 'ba2a48c7d50a57f01798aab3e7162597',
    'x-dp-watson-tran-id': 'ba2a48c7d50a57f01798aab3e7162597',
    'x-edgeconnect-midmile-rtt': '16',
    'x-edgeconnect-origin-mex-latency': '323',
    date: 'Tue, 17 Nov 2020 21:13:52 GMT',
    connection: 'close'
  }
}

jweisz, 2020-11-18:
Hi @MUbarak123-56, could you try these two things?

1. Please verify your config.js looks like the following, to make sure you put the assistantId in the right place. All the 'x's should be where you put your assistantId.

```
export default {
    assistantId: 'xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx', // add your assistant id from Watson Assistant
    hasCamera: true, // set this to false if your TJBot doesn't have a camera
    robotName: 'Watson', // set this to the name you wish to use to address your tjbot!
};
```

2. Please set the logging level to `silly` and run the recipe again.

```
const tjConfig = {
    log: {
        level: 'silly', // change to 'verbose' or 'silly' for more detail about what TJBot is doing
    },
    ...
```

Look for a line that starts with `response from _assistant.createSession()`, which should have more detail as to what is failing.

MUbarak123-56, 2020-11-18:
the assistantId is in the right place but I'll try it again when I get to work

MUbarak123-56, 2020-11-18:
@jweisz  I did what you said with the silly stuff but it just returned the same error:

info: Hello from TJBot!
verbose: TJBot library version v2.0.0
silly: TJBot configuration: {"log":{"level":"silly"},"robot":{"gender":"male"},"converse":{"assistantId":"a7d3f382-d76b-43a1-8b56-15e53798e13f"},"listen":{"microphoneDeviceId":"plughw:1,0","inactivityTimeout":-1,"backgroundAudioSuppression":0.4,"language":"en-US"},"wave":{"servoPin":7},"speak":{"language":"en-US","speakerDeviceId":"plughw:0,0"},"see":{"confidenceThreshold":0.6,"camera":{"height":720,"width":960,"verticalFlip":false,"horizontalFlip":false}},"shine":{"neopixel":{"gpioPin":18,"grbFormat":false},"commonAnode":{"redPin":19,"greenPin":13,"bluePin":12}}}
info: Initializing TJBot with microphone,speaker,led_neopixel,servo,camera
verbose: initializing microphone
verbose: initializing speaker
verbose: initializing led_neopixel on PIN 18
verbose: initializing servo on PIN 7
verbose: initializing camera
You can ask me to introduce myself or tell you a joke.
Try saying, "Watson, please introduce yourself" or "Watson, what can you do?"
You can also say, "Watson, tell me a joke!"
Say 'stop' or press ctrl-c to exit this recipe.
verbose: initializing speech_to_text service
verbose: initializing microphone
silly: recognizeUsingWebSocket() params: {"objectMode":false,"contentType":"audio/l16; rate=16000; channels=1","model":"en-US_BroadbandModel","inactivityTimeout":-1,"interimResults":true,"backgroundAudioSuppression":0.4}
verbose: microphone started
verbose: microphone silence
info: TJBot heard: "Watson hello"
verbose: initializing assistant service
error: error creating session for assistant service. please check that tj.configuration.converse.assistantId is defined.
/home/pi/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/request-wrapper.js:224
        var error = new Error();
                    ^

Not Found: Resource not found
    at RequestWrapper.formatError (/home/pi/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/request-wrapper.js:224:21)
    at /home/pi/tjbot/recipes/conversation/node_modules/ibm-cloud-sdk-core/lib/request-wrapper.js:212:25
    at processTicksAndRejections (node:internal/process/task_queues:93:5) {
  statusText: 'Not Found',
  status: 404,
  code: 404,
  body: '{"error":"Resource not found","code":404}',
  headers: {
    'content-type': 'application/json; charset=utf-8',
    'content-length': '41',
    'access-control-allow-origin': '*',
    'access-control-allow-methods': 'GET, POST, PUT, DELETE, PATCH, HEAD, OPTIONS',
    'access-control-allow-headers': 'Content-Type, Content-Length, Authorization, X-Watson-Authorization-Token, X-WDC-PL-OPT-OUT, X-Watson-UserInfo, X-Watson-Learning-Opt-Out, X-Watson-Metadata',
    'access-control-max-age': '3600',
    'content-security-policy': "default-src 'none'",
    'x-dns-prefetch-control': 'off',
    'x-frame-options': 'SAMEORIGIN',
    'strict-transport-security': 'max-age=31536000; includeSubDomains;',
    'x-download-options': 'noopen',
    'x-content-type-options': 'nosniff',
    'x-xss-protection': '1; mode=block',
    'x-response-time': '12.636ms',
    'x-global-transaction-id': '1bad28e8279e95b834ffa7249927216a',
    'x-dp-watson-tran-id': '1bad28e8279e95b834ffa7249927216a',
    'x-edgeconnect-midmile-rtt': '21',
    'x-edgeconnect-origin-mex-latency': '155',
    date: 'Wed, 18 Nov 2020 16:31:41 GMT',
    connection: 'close'
  }
}

MUbarak123-56, 2020-11-20:
@jweisz Hi, please can you help me fix this error?

tavaresft, 2020-12-02:
I'm similar error, please Someone help us? This issue only happen when I execute in raspberry, if I using simulator (https://my-tjbot.mybluemix.net/) it's not happens

tavaresft, 2020-12-09:
I am not using the assistant (Watson API) from this library (tjbot), because of this bug. So, for now I used the tutorial below, to use assistant API.

For more information to implementing Watson Assistance, follow:
https://cloud.ibm.com/docs/assistant?topic=assistant-api-client&locale=pt-BR&programming_language=javascript

MUbarak123-56, 2020-12-09:
@tavaresft Thank you

jweisz, 2021-01-15:
Hi, sorry for the delay in responding. I'm actually stumped by this, I'm not sure why Watson Assistant API should be returning a 404 Not Found when trying to create the session. I have a few theories / suggestions:
- there may have been a behind-the-scenes update to the Assistant service that prevents the latest `ibm-watson` SDK from working properly. to see if this is the case, try creating a new instance of Assistant and creating a new workspace from scratch, then seeing if the error happens in that new workspace.
- the `ASSISTANT_URL` parameter might be missing from your `ibm-credentials.env` file, though I suspect this isn't the problem

One way to try to debug this is to edit the file `node_modules/ibm-cloud-sdk-core/lib/request-wrapper.js`, go to line 194, and insert the following on the next line:

```
console.log('request' + JSON.stringify(requestParams));
```

That should print out the entire contents of the request tjbot is making when it tries to create an Assistant session. Maybe there's a clue in there as to what's going wrong.

MUbarak123-56, 2021-01-15:
> Hi, sorry for the delay in responding. I'm actually stumped by this, I'm not sure why Watson Assistant API should be returning a 404 Not Found when trying to create the session. I have a few [...]

Thank you very much for the response

MUbarak123-56, 2021-02-01:
> Hi, sorry for the delay in responding. I'm actually stumped by this, I'm not sure why Watson Assistant API should be returning a 404 Not Found when trying to create the session. I have a few [...]

Hello I followed up on your feedback and used the console.log('request: ' + JSON.stringify(requestParams)); but it did not work. Are there any updates for fixing it.

MUbarak123-56, 2021-02-05:
@jweisz @jweisz I tried out installing everything from scratch using the 2.0.1 version and running the entire code all over again. However, I got the same response again when I ran sude node conversation.js. It keeps reporting an error called var error = New Error();. I am quite confused about where to move from here.

CaskAle, 2021-02-05:
What version of node.js and npm are you running.
`node -v`
`npm -v`

I think these new libraries require at least v12 of node.js

MUbarak123-56, 2021-02-05:
@CaskAle I am using v15.8.0 for node.js

MUbarak123-56, 2021-02-26:
@CaskAle @jweisz Have you found any solutions yet for the bug?

jweisz, 2021-02-26:
@MUbarak123-56 

> Hello I followed up on your feedback and used the console.log('request: ' + JSON.stringify(requestParams)); but it did not work. Are there any updates for fixing it.

What does "it did not work" mean? That line of code wasn't going to fix your problem, but it might have provided useful diagnostic information by showing exactly what request your Pi was sending to the Watson service. Essentially, the problem is that this line is returning an error in `tjbot.js:734`:

```
const body = await this._assistant.createSession({
                    assistantId: this.configuration.converse.assistantId,
                });
```

The only reasons I can think of this failing (because all it's doing is calling into the Watson Assistant service.. has nothing to do with TJBot) are because:
1. `this.configuration.converse.assistantId` is not defined
2. there is something misconfigured with your instance of the Assistant service (e.g. no assistant with that id defined)
3. your instance of the Assistant service may not work with the version of the `ibm-watson` SDK being used by TJBot (`5.7.0`, which is relatively new)

It sounds like the issue isn't (1), and I thought you had tried creating a new instance of Assistant which would rule out 2 and 3. But if you haven't tried that yet, that would be my next recommendation.

MUbarak123-56, 2021-02-27:
@jweisz I already tried out other assistant instances but they did not seem to work. I tried out the stt.js module and that works perfectly well. It's just the conversation.js that is failing to work right now. My colleagues who recently built their TJbot from scratch and created new assistances are also facing the same issues. Please, is it possible for us to set up a virtual meeting in which you assist me solve the problem?

CaskAle, 2021-02-27:
Well, I now cant get past an STT error during microphone init.  It worked last time i tried it.  I will need to get past this before I am able to assist.  My only potential suggestion is to make sure that you are using the correct assistant ID.  I had some students inadvertently using an api key here.  Once they corrected it, all was well.

MUbarak123-56, 2021-02-27:
@CaskAle Hi, How is it going? I already corrected mine multiple times and created new instances but the assistantID does not seem to work. Please can I set up a zoom meeting with you so you can guide me on how to solve the problem?

MUbarak123-56, 2021-03-05:
@jweisz Hello, I finally figured out my mistakes and I got the TJbot to work. First of all, my Watson assistance service on the IBM cloud was outdated so I had to delete it and create a new one. Secondly, I was using the skillID in place of the assistantID. I did not realize that the new version required the 'actual' assistantID.  Sorry for bothering you too much. I was also wondering if I could have a virtual chat with you some time about running other projects centered around TJbot.

closed 2021-03-05 by MUbarak123-56