ibmtjbot/tjbot #176: TJBot & raspi 4b - running conversation stops and kills audio via jack

juliaheller, 2021-02-11:
When I first installed TJBot and disabled the audio jack I could make it running but didn't get audio (via bluetooth). 

I installed it a second time without disabling the audio jack during installation. Now when I run sudo node conversation.js it suddenly stops after the microphone started - without error. What also happend: I listened to music on youtube with headphones via the audio jack. The sound broke when I ran conversation.js. Only got annoying high noices.
Had to reboot to have sound again.

jweisz, 2021-02-11:
Hi @juliaheller, sorry to hear about your audio problems. Have you tried using a wired speaker? We've had a lot of issues with bluetooth speakers and generally don't recommend them. We recommend using a USB audio adapter (e.g. https://www.adafruit.com/product/1475) to avoid the hassles of bluetooth, as well as the noise from the 3.5mm audio jack when using LEDs.

juliaheller, 2021-02-12:
@jweisz thanx for your answer. I installed TJBot again and this time disabled the audio jack again. Now it's running again. I have an USB speaker but so far I couldn't get a sound. I ordered an audio adapter. Hope this'll do. Thanx again!

closed 2021-02-12 by juliaheller

CaskAle, 2021-02-12:
When using a USB speaker, you may need to adjust the device that you wish the sound to go to.  By default, sound goes out to audio card 0 and you may need to change that to card 1.  Have you tried this already?

juliaheller, 2021-02-12:
No, I haven’t. Will try it tonight. Thanx a lot!