ibmtjbot/tjbot #179: Update instructables to mention how to input apikeys

robertoetcheverryr, 2021-10-10:
Hi, I had to read and debug some of the code to finally understand why, even though I followed the instructions (for the conversation recipe) that said to create my services and input the assistant ID into the config.js, the recipe still complained about not having the credentials for the service. sometime in the tjbot v2 migration they got moved to the ibm-credentials.env file, but there is no mention of how to populate (apikey vs iam apikey, fields where you apparently don't need to put anything, etc).

heather-fillerup-miller, 2022-12-08:
We could update the ibm-credentials.sample.env and README.env to include more detailed instructions on how to get the appropriate credentials.

jweisz, 2023-01-27:
We need to decide where @ljmwaugh 's new instructions will live: on the GitHub pages site, or as a new Instructable (or both?). Low priority until we figure out a few more things…