ibmtjbot/tjbot #180: Remove support for camera due to Visual Recognition API not being available anymore

robertoetcheverryr, 2021-10-10:
As the title says, May I make a PR removing the config.js option for the camera or at least adding a comment that it's not available anymore? And the same with the ibm-credentials.env file? Maybe delete the switch case regarding the camera?

robertoetcheverryr, 2021-10-22:
I haven't used it yet, but Google has a Vision API with 1000 monthly queries. I'm not sure if that works with the free account or with the "you must put your CC and only then you get the free tier"...

andycitron, 2021-12-13:
I do have an already-provisioned visual recognition service provisioned, but it seems to have stopped working.  I'm having trouble figuring out why.  The reply tjbot returns says:  Error: <HTML><HEAD><TITLE>Error</TITLE></HEAD><BODY>
An error occurred while processing your request.<p>
Reference&#32;&#35;30&#46;713a2f17&#46;1639419281&#46;1a1a60f6
</BODY></HTML>
 Do I have any hope of working around this?

andycitron, 2021-12-24:
Justin, I reimplemented my tjbot code using Microsoft Azure visual services.  I uploaded the code fragments to github in case you were interested in incorporating it into your tjbot node js implementation.  You can find it here: 
https://github.com/andycitron/tjBotFragmentThatUsesAzureVisualServices

Note that it does introduce a dependency that the user has a Microsoft Azure account.  

Also, if you want to incorporate it into tj.see(), you'd want to structure it a bit differently.  tj.see() takes a photo.  I did not want my Microsoft functions to have a dependency on tj.takePhoto.  So I put the 'take a photo' part into the intent processing for 'see' and passed the photo into the code that uses Microsoft functions.

The code I put out there includes additional methods that invoke Microsoft facial recognition.  That is not part of the 'tj.see()' functionality.  That code requires pre-training of the facial recognition models.   I included that because it might be useful to someone.

jweisz, 2022-01-04:
Hey @andycitron, happy new year. :) 

Thanks for the effort you put into TJBot, this is a really great contribution. Unfortunately, I won't be able to make this part of the official library because it uses a competitor's cloud service. But, I will put this on our Featured Recipes page to showcase your work.

andycitron, 2022-01-05:
Cool.  Yesterday I posted a video to Youtube illustrating how TJ works with Azure.  Perhaps you want to include the 4 minute video along with the featured recipe:  https://youtu.be/B92efwFqXSs 

Could you give me the link to the Featured Recipes page where you referenced my code?  I'd like to include a link to that on my home page.  thx.

jweisz, 2022-01-05:
Here's the link: https://github.com/ibmtjbot/tjbot/tree/master/featured#microsoft-azure-visual-services-by-andycitron

andycitron, 2022-05-30:
Justin,
Sorry to bother you, but I can’t figure out where to ask this question.  It’s not a tjbot issue, just a question.

Where does tjbot store the audio file it uses for speech to text?  What format?

I see that Microsoft Azure has ‘person voice recognition’ and I’m thinking about trying that out.  Seems it wants a wav audio file as input.

My tjbot gets confused when multiple speakers are talking at the same time.   Those utterances usually end up being ignored by my implementation.  But every once in a while it’ll try to respond.  Because I know who I’m talking to (facial recognition) I can ignore utterances from a different person….or at least I can try.

Do you know the answer?  Or can you tell me the proper place to ask this.

I actually think an implementation using multiple microphones and detecting voice based on location in the room makes sense, but that seems very hard.

Thx,
Andy

jweisz, 2022-05-31:
Hi @andycitron -- take a look at `tjbot.js:792`, where `listen()` is defined:
https://github.com/ibmtjbot/tjbotlib/blob/4fe0263bd0050f910752ae589d3b33cdb9cb93ae/src/tjbot.js#L792

The audio isn't stored locally, the data is streamed through a pipe between the microphone and a web socket. So it would need some modification to save the output to a file first, before uploading to the Microsoft service. Maybe check to see if their API supports WebSockets?

jweisz, 2023-01-13:
Closing as this is now an issue in the tjbotlib repo: https://github.com/ibmtjbot/tjbotlib/issues/73

closed 2023-01-13 by jweisz