ibmtjbot/tjbot #181: Conversation recipe broken due to Watson Assistant defaults

robertoetcheverryr, 2021-10-10:
The conversation recipe seems to assume that a freshly created Watson Assistant will have some intents and entities defined (greeting, shine, wave, see, colors for the LED, etc).
But nowadays a freshly created Watson Assistant only has one intent and it doesn't match any of the example expressions used by the recipe.
Besides reading the code, is there any reference to how the Assistant should be configured?

CaskAle, 2021-10-11:
From the README:
Set up Watson Assistant using the following steps:

Launch the Watson Assistant tool and create a new assistant.
Click "Add dialog skill" and then "Import Skill."
Upload the tjbot-skill-sample.json file.
Go back to the Assistants screen and click the menu (with the three dots), and click "Settings."
Click "API Details" in the left sidebar.
Copy the "Assistant ID" and paste it into your config.js file
Run!

robertoetcheverryr, 2021-10-11:
@CaskAle Distributed readmes do not help it seems. Either that readme is outdated (as mentioned on another Issue) and doesn't tell you to input your apikeys and etc into the .env file or it's a leftover from before the 2.0 migration.
But yes. I overlooked that part. Thanks!