ibmtjbot/tjbot #211: Raspberry Pi 4 - Neopixel LED Initialization Errors

heather-fillerup-miller, 2023-01-10:
When trying to run the led-test and speech_to_text recipe, the following error occurs:

error initializing - 3 Hardware revision is not supported
Verbose: shining my LED to RGB color 0xFF0000
Segmentation fault

Found a temporary solution here: https://github.com/beyondscreen/node-rpi-ws281x-native/issues/130

heather-fillerup-miller, 2023-03-10:
Turns out we have addressed this issue, we have a comment about it in the TROUBLESHOOTING.md file under "LED doesn't work on Raspberry Pi 4".

I will test that solution to ensure it still works and if so update the documentation so that it also mentions the "error initializing - 3 Hardware revision is not supported" and will be easier to find.

jweisz, 2023-04-07:
- [ ] Need to test the LED solution that's in TROUBLESHOOTING to make sure it still works.
- [ ] Need to re-word TROUBLESHOOTING
- [ ] Possibly explore whether we should use a different library for driving the LED instead of `rpi-ws281x`

jweisz, 2023-09-22:
@heather-fillerup-miller will double-check the status of the library to see if the PR was accepted that fixes RPi4 support for LEDs

heather-fillerup-miller, 2023-09-30:
Does not look like the PR has been accepted yet: https://github.com/beyondscreen/node-rpi-ws281x-native/pull/134

closed 2023-10-06 by heather-fillerup-miller