ibmtjbot/tjbot #228: Verify that hardware tests and recipes work with RP4

jweisz, 2023-04-07:
There appear to be several issues when attempting to set up TJBot on RP4 so we need to verify that it works properly.

jweisz, 2023-04-07:
Two part plan:
1. get USB audio working, figure out an easy way for people to configure the hw device (e.g. the `plughw0,1` thing)
2. also looking into this: https://learn.adafruit.com/adafruit-speaker-bonnet-for-raspberry-pi/overview and write some instructions for how to set it up

jweisz, 2024-03-29:
Should also test the bootstrap script on the new OS (bookworm): https://www.raspberrypi.com/software/operating-systems/

jweisz, 2024-03-29:
@heather-fillerup-miller may want to get a USB sound card for RPi 3 testing: https://www.adafruit.com/product/1475

andypiper, 2024-04-29:
I just hit "the plughw0,1 thing" I think - updated Raspberry Pi OS on my TJBot and it is reporting issues at boot with the GEMBIRD Honk HK-5002 USB Speaker and I can only get it to make a short noise when playing a wav. Will have a poke when I have more time.

jweisz, 2024-04-29:
@andypiper in case this is helpful, I try to diagnose audio issues by using `aplay -l` to list out my audio devices and then playing this sound file:

```
$ aplay -D plughw:0,0 /usr/share/sounds/alsa/Front_Center.wav
```

where the numbers in `plughw` come from careful scrutiny of `aplay -l` 😀 

I'm not specifically familiar with the GEMBIRD though.

jweisz, 2024-10-11:
@heather-fillerup-miller will bootstrap an RPi 4 and try running the hardware tests + the new Watsonx.ai recipe.