ibmtjbot/tjbot #231: Migrate instructions content to the GitHub wiki

jweisz, 2023-04-21:


jweisz, 2023-05-05:
Here are the new hardware setup instructions: [Watson Services in IBM Cloud v3.docx](https://github.com/ibmtjbot/tjbot/files/10415863/Watson.Services.in.IBM.Cloud.v3.docx]

jweisz, 2023-05-05:
The wiki is in progress! And we have a very cool cover image :D

jweisz, 2023-06-16:
Feedback from June 16:

- Can we fix the order of the links in the sidebar?
-   About TJBot
-   Build TJBot
-   Bring TJBot to life
-   Recipes (or "Official Recipes")
-   Community Recipes (rename it from "Featured Recipes")
-   Testing
-   Troubleshooting (maybe merge with Testing?)
-   Tutorials (maybe merge with Bring TJBot to Life?)
-   Contribute

- Need to add Lisa's material on provisioning cloud services

heather-fillerup-miller, 2023-06-29:
The default behavior for the sidebar topics is alphabetical, but I believe there is a way to customize it, so I will look into. 

I agree with the renames and merging Tutorials into Bring TJBot to life, but maybe we should merge Testing into Troubleshooting rather than the other way around?

I am also working with Lisa to get her cloud service materials added. We ran into an issue with the markdown conversion of the images from Word.

jweisz, 2023-06-29:
Testing feels like a thing everyone should do after building their TJBot. Troubleshooting feels like the place you go when you run into problems at any phase. So maybe these really are separate?

heather-fillerup-miller, 2023-06-29:
I think it makes sense to keep the testing separate from troubleshooting since it is something that should be done before using the recipes to ensure that the hardware is set up and working properly.

My plan was to add the README content from the tests folder so this section has a link to the tests folder and explains how to run the tests. 

Maybe we rename it to "TJBot Hardware Tests"? That way it is more descriptive?

jweisz, 2023-06-30:
"TJBot Hardware Tests" or "Hardware Tests" sounds good. Thanks @heather-fillerup-miller !

heather-fillerup-miller, 2023-09-30:
Added Lisa's Watson Services section and converted all images from inches to pixels. Renamed sections based on discussion above. Currently order of sections is alphabetical, but looking into how to customize order.

heather-fillerup-miller, 2023-10-06:
Found issues with links in Recipes, need to clean up and link to READMEs

heather-fillerup-miller, 2023-10-17:
Made updates to wiki, links seems to working appropriately now, the issue was when I created a custom side bar, it broke the internal linking to the markdown files in the wiki repo.

jweisz, 2024-03-29:
@heather-fillerup-miller @jweisz let's take a pass through the wiki content and clean up language like "TJBot is optimized for RPi 3" since that's not exactly true anymore :)