ibmtjbot/tjbot #250: Rename `conversation` recipe to `conversation-assistant`

jweisz, 2023-09-22:
The existing `conversation` recipe uses Watson Assistant, but WA has likely changed a lot since we wrote the recipe (there's a new thing called "Actions" now?), so the proposal is to:

1. rename the recipe to make it clear that it is demonstrating conversation using Watson Assistant (hence, `conversation-assistant`)
2. update the recipe to work with the new way that WA conversations get defined. I'm not sure the existing `.json` file that serialized all our conversational intents will work with the new WA? This needs to be tested. 😄

modcarroll, 2024-02-27:
Working on updating the .json now. Question for this one though: the new proper name for conversation is "watsonx Assistant" so should our naming reflect that? `watsonx_assistant`

modcarroll, 2024-02-27:
Actually in thinking this through a bit more, I take that back because there are two recipes we could go with for assistant:
1. Use watsonx to generate answers
2. Re-use our existing sample .json

So maybe `watsonx_assistant` for 1, then `conversation_assistant` for 2?

jweisz, 2024-02-28:
Hm, good question. I guess the way I thought about it was the way you described above:

1. Use watsonx.ai to directly generate responses to things people speak to TJBot. This is probably the most interesting use case right now given the excitement about LLMs.
2. Use watsonx Assistant to show how to build a dialogue model for TJBot. It's been a long time since I used assistant and I'm not even sure if our `.json` file is still valid, although I think I did do some fixup at one point (e.g. to remove functionality we no longer have)

If I had to prioritize, 1 is at the top of my list. :)

Technically this issue should probably be closed given that we did already rename the receipts (in `dev`): (1) corresponds to `conversation_watsonx` and (2) corresponds to `conversation_assistant`. And I think we have other issues for actually implementing these recipes? I'll let you decide @modcarroll :D

PR #264 (2024-03-04): Moves assistant API calls into the recipe
Resolves #259 
Resolves #250
closed 2024-03-04 by modcarroll